open-rmf/rmf_ros2 #9: Allow navigation graph lanes to be disabled

mxgrey, 2021-03-15:
## Feature request

Allow lanes to be disabled

### Description

There may be various reasons that the lane of a navigation graph needs to be disabled. If a corridor is needs to be closed off for maintenance, or if too many obstacles have been detected, we should be able to tell the fleet adapter to disable the lane and plan around it.

txlei, 2021-03-16:
@mxgrey would there be a method to "enable" the lane again?

Also, it was decided that regardless of the obstacle detected, the robot would attempt to overcome it using its local planner. When it concludes its attempt as a failure, the replanning would be last resort. Hence, the robot is already on the "impassable lane". Would that implicate rmf's global planner efforts to replan?

mxgrey, 2021-03-16:
> would there be a method to "enable" the lane again?

Yes, absolutely. This feature will allow lanes to be toggled on and off.

> the robot is already on the "impassable lane". Would that implicate rmf's global planner efforts to replan?

No worries, I fully intend to handle this situation. I expect this will be a common occurrence :+1:

txlei, 2021-03-19:
> disable the lane and plan around it.

Does this API require location of the obstacle as input? Or it assumes the obstacle is somewhere in the same direction of where the robot is heading towards. i.e. On the traffic lane, the path in front of the robot is impassable and the path behind the robot is passable.

mxgrey, 2021-03-19:
Nope, the API will just specify which lane(s) to close down. From the planner's point of view, the only thing it means if a lane is closed is that the planner can't expand its path search down that lane.

I can put some logic into the fleet adapter that will check if the robot is currently on a lane that's being closed, and then the fleet adapter will have the robot return to the lane's entry waypoint if it is.

txlei, 2021-03-19:
That works. Thank you.

mxgrey, 2021-03-25:
Finished by #15

closed 2021-03-25 by mxgrey