open-rmf/rmf_ros2 #20: restart error Failed to correctly load participant registry

txlei, 2021-04-01:
Hi, I am not sure why this error happens after restarting the office demo. I've done a PC restart, rosdep update and rebuild. Currently doing a clean rebuild. 
```
[rmf_traffic_schedule-1] [FATAL] [1617249496.967795104] [rmf_traffic_schedule_node]: Failed to correctly load participant registry: yaml-cpp: error at line 5208, column 5: Participant description missing [profile] field
[rmf_traffic_schedule-1] 
[rmf_traffic_schedule-1] terminate called after throwing an instance of 'std::runtime_error'
[rmf_traffic_schedule-1]   what():  yaml-cpp: error at line 5208, column 5: Participant description missing [profile] field
[ERROR] [rmf_traffic_schedule-1]: process has died [pid 42702, exit code -6, cmd '/home/leia/rmf_demos_ws/install/rmf_traffic_ros2/lib/rmf_traffic_ros2/rmf_traffic_schedule --ros-args --params-file /tmp/launch_params_ttpvadv4'].
[rviz2-7] [INFO] [1617249497.414677664] [door_panel_requester_node]: New door [hardware_door] found, refreshing...
[rviz2-7] [INFO] [1617249497.416391942] [door_panel_requester_node]: New door [main_door] found, refreshing...
[rviz2-7] [INFO] [1617249497.632252148] [door_panel_requester_node]: New door [coe_door] found, refreshing...
[ERROR] [full_control-15]: process has died [pid 42730, exit code 1, cmd '/home/leia/rmf_demos_ws/install/rmf_fleet_adapter/lib/rmf_fleet_adapter/full_control --ros-args -r __node:=tinyRobot_fleet_adapter --params-file /tmp/launch_params_kky8_qz0 --params-file /tmp/launch_params_ot2362qo --params-file /tmp/launch_params_nbxp0qgd --params-file /tmp/launch_params_11bjqwnr --params-file /tmp/launch_params_ga4pffd1 --params-file /tmp/launch_params_owr6ktwe --params-file /tmp/launch_params_z9352fdz --params-file /tmp/launch_params_yl18cv02 --params-file /tmp/launch_params_4ykopwp0 --params-file /tmp/launch_params_gjljm02w --params-file /tmp/launch_params_1x2psn48 --params-file /tmp/launch_params_wkv0a8nd --params-file /tmp/launch_params_7n2g8jt7 --params-file /tmp/launch_params_0501cpoa --params-file /tmp/launch_params_tw9o2d7h --params-file /tmp/launch_params_gbkq9dyl --params-file /tmp/launch_params_s0z9jzgo --params-file /tmp/launch_params_gts2bu7v --params-file /tmp/launch_params_af5tqyi4 --params-file /tmp/launch_params_evo7ghis --params-file /tmp/launch_params_zknsaww7 --params-file /tmp/launch_params_zdq4nt43 --params-file /tmp/launch_params_nrslp4yi --params-file /tmp/launch_params_kuy3pllb --params-file /tmp/launch_params_84u8l3mk --params-file /tmp/launch_params_hb4u79mh --params-file /tmp/launch_params_gv_ba95g --params-file /tmp/launch_params_iicc1evb'].
```

Yadunund, 2021-04-01:
Could you try deleting the `.rmf_schedule_node.yaml` that is generated in the directory from which you launch the simulation? For example if you launch `ros2 launch rmf_demos office.launch.xml` from `rmf_demos_ws/` then `rm rmf_demos_ws/.rmf_schedule_node.yaml`. Re-launching the simulation should solve the issue.

mxgrey, 2021-04-01:
It looks like we forgot to have the schedule node print out what log file it tried to open if the parsing fails. We should add that information to [this line](https://github.com/open-rmf/rmf_ros2/blob/2fe08e328f543fe6a4e0853a60607b5b52015f2a/rmf_traffic_ros2/src/rmf_traffic_ros2/schedule/Node.cpp#L113). Pointing users to the relevant file should help with this kind of situation.

What surprises me is that there'd be something wrong with that file in the first place. @txlei would you be able to share the current state of the `.rmf_schedule_node.yaml` file with us before you delete it?

txlei, 2021-04-01:
@mxgrey failed to recover it. My drives are kinda iffy because mine's a multi-boot machine. 

It never happened before only twice today. Would keep you posted if it happens again

closed 2021-04-01 by txlei