open-rmf/rmf_ros2 #28: Hardcoded timeout values may be too strict for deployment scenarios

cnboonhan, 2021-04-12:
## Bug report
Hardcoded timeout values, such as in https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/full_control/main.cpp#L350, seem too strict, or might be better set as a parameter for deployment scenarios.
 
### Required information:

- Operating system and version:
  - 
- OpenRMF installation type:
Built from source
- OpenRMF version or commit hash
master
- ROS distribution and version:
Foxy
- ROS installation type:
  Binaries

### Description of the bug
When running RMF over a cloud network, the hardcoded timeouts for determining if a path request got dropped is written as 200 milliseconds. In a physical deployment, this might be too strict. A setup that I made would, as a result, endlessly get Path Requests resent as the network fails to meet this latency requirement.

I have checked that time synchronization is active (`timedatectl set-ntp true`) but pings to the cloud server was around 40ms, but can spike.

### Steps to reproduce the bug
This is quite complex to describe, but my short fix solution was to increase this timeout and rebuild the code. This seems to suggested a two pronged approach of stricter time synchronization infrastructure and a parameterized option for such timeouts be useful.

mxgrey, 2021-04-12:
I don't mind creating a parameter for this value, but is resending the command at a rate of 5Hz problematic?

cnboonhan, 2021-04-12:
The issue for me seemed to me the following on free fleet ( as i'm using the free fleet implementation )
1. Robot is instructed to move to destination A
2. Robot arrives at destination A
3. Robot continues to be instructed to move to destination A at 5Hz
4. This interrupts any further assignments from the dispatcher to this robot ( web interface )

The result is that the robot does not receive any further tasks assignments.

mxgrey, 2021-04-12:
I think there must be some other problem happening, and this republishing is only a symptom of it rather than the cause.

If you look at the [parent condition](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/full_control/main.cpp#L346) this republishing will only happen if the latest state that the fleet adapter receives from the robot has the wrong `task_id`. When that happens, the fleet adapter (quite reasonably) assumes that the robot has not received its intended command, so the command gets resent.

If the robot is making it all the way to its destination while the fleet adapter has the wrong `task_id` for it, then I imagine something is interfering with the communication from the robot to the fleet adapter.

Also, this command republishing does not interfere with the ability to receive more task assignments. If a new task assignment comes in, the fleet adapter will simply switch over to the new task. The only thing that would interfere with starting a new task is if the fleet adapter believes that the robot has not completed its current task yet, which again would be related to the comms from the robot to the fleet adapter.

cnboonhan, 2021-04-12:
that's a great point. I'll have to try this again and observe the logs when I next can.

The first task my robot gets is an "automatic" one, that the RMF backend sends to move the robot to the nearest graph waypoint. It's hard to imagine what other task_id could be getting in the way.

mxgrey, 2021-04-12:
Did you find that increasing the timeout actually helped with the issue? If so, I'd be quite perplexed as to why.

cnboonhan, 2021-04-12:
It seemed to, when i increased it to (i believe) 2000ms, from 200ms. 
I should do more concrete experiments on this since I had to make various changes in many areas that might confuse my findings.

cnboonhan, 2021-04-14:
Upon further analysis it seems like the issue was a combination of misunderstanding the expected behavior of ResponsiveWait devices the effects of the incorrect use of `use_sim_time` and the importance of synchronized time on all devices involved in the deployment. I'm able to run a deployment without having to change this parameter, so I'll close this issue. Thanks!

closed 2021-04-14 by cnboonhan