open-rmf/rmf_ros2 #33: Can't find `pybind11_vendor` package

aaronchongth, 2021-04-16:
## Bug report

### Required information:

- Operating system and version:
  -  Ubuntu 20.04
- OpenRMF installation type:
  - Source build
- OpenRMF version or commit hash
  - https://github.com/open-rmf/rmf_ros2/pull/32
- ROS distribution and version:
  - Foxy
- ROS installation type:
  - Binaries
- Package or library, if applicable:
  - `rmf_fleet_adapter_python`

### Description of the bug

Fails to find package `pybind11_vendor` while building `rmf_fleet_adapter_python`,

```
--- stderr: rmf_fleet_adapter_python                                       
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "Findpybind11_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "pybind11_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "pybind11_vendor"
  with any of the following names:

    pybind11_vendorConfig.cmake
    pybind11_vendor-config.cmake

  Add the installation prefix of "pybind11_vendor" to CMAKE_PREFIX_PATH or
  set "pybind11_vendor_DIR" to a directory containing one of the above files.
  If "pybind11_vendor" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< rmf_fleet_adapter_python [5.70s, exited with code 1]
```

### Steps to reproduce the bug

1. Build up to `rmf_fleet_adapter_python`

```
mkdir -p ~/rmf_ws/src
cd ~/rmf_ws
wget https://raw.githubusercontent.com/open-rmf/rmf/main/rmf.repos
vcs import src < rmf.repos

source /opt/ros/foxy/setup.bash
colcon build --cmake-args -DCMAKE_BUILD_TYPE=RELEASE -DBUILD_TESTING=OFF --packages-up-to rmf_fleet_adapter_python
```

ddengster, 2021-04-16:
i think they fixed a similar issue a while ago,  do the standard `rosdep update` and `rosdep install something`or install the missing package via `apt-get install ros-foxy-pybind11-vendor`

aaronchongth, 2021-04-16:
`rosdep_update` did the trick, thanks!

closed 2021-04-16 by aaronchongth