open-rmf/rmf_ros2 #38: Provide infrastructure integration for Traffic Light API

mxgrey, 2021-04-22:
Migrated from https://github.com/osrf/rmf_core/issues/239

kris24x10, 2022-08-04:
Hi! Just wanted to check whether there has been any update on this issue?

mxgrey, 2022-08-04:
We don't presently have a funding source for this feature since most development interest has been around enhancing the full control and eventually native nav stack integration.

That being said, this feature is really just a quality of life improvement for system integrators, not a blocker for any system capabilities. There's nothing preventing a Traffic Light API user from manually integrating with the door+lift+dispenser messages within their fleet adapter implementation code.

kris24x10, 2022-08-04:
Noted, thanks Grey for the detailed reply!