open-rmf/rmf_ros2 #62: task type and other fields not populated correctly in "/task_summaries" topic

koonpeng, 2021-06-01:
## Bug report

Tasks published in `task_summaries` all have task type `0` (station task type), it also publishes an empty string for other fields like `start_name` and `finish_name`, it seems like the whole `description` is being published with a zero initialised value. An example of a loop task being published as a station task:

```yaml
fleet_name: tinyRobot
task_id: Loop0
task_profile:
  task_id: Loop0
  submission_time:
    sec: 0
    nanosec: 0
  description:
    start_time:
      sec: 0
      nanosec: 0
    priority:
      value: 0
    task_type:
      type: 0
    station:
      task_id: ''
      robot_type: ''
      place_name: ''
    loop:
      task_id: ''
      robot_type: ''
      num_loops: 0
      start_name: ''
      finish_name: ''
    delivery:
      task_id: ''
      items: []
      pickup_place_name: ''
      pickup_dispenser: ''
      pickup_behavior:
        name: ''
        parameters: []
      dropoff_place_name: ''
      dropoff_ingestor: ''
      dropoff_behavior:
        name: ''
        parameters: []
    clean:
      start_waypoint: ''
state: 0
status: 'Finding a plan for [tinyRobot/tinyRobot2] to go to [4] | Remaining phases: 2'
submission_time:
  sec: 0
  nanosec: 0
start_time:
  sec: 40
  nanosec: 52000000
end_time:
  sec: 123
  nanosec: 823499496
robot_name: tinyRobot2
```

### Required information:

- Operating system and version:
  - ubuntu 20.04
- OpenRMF installation type:
  - soruces
- OpenRMF version or commit hash
  - main
- ROS distribution and version:
  - foxy
- ROS installation type:
  - binaries
- Package or library, if applicable:
  - 

### Description of the bug

### Steps to reproduce the bug

1. Listen to `/task_summaries` topic
1. Create a loop task in the web dashboard

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

Yadunund, 2021-06-01:
PR https://github.com/open-rmf/rmf_ros2/pull/24 addresses this bug but is still under review. In the meanwhile for your testing, you can use the `throwaway/demo` branch which correctly populates the task_type and relevant fields in the TaskSummary message.

Yadunund, 2022-05-12:
Closing since all task states and logs are published over websocket as of https://github.com/open-rmf/rmf_ros2/pull/168

closed 2022-05-12 by Yadunund