open-rmf/rmf_ros2 #74: __init__() got an unexpected keyword argument 'noninteractive'

FerusOlin, 2021-07-06:
## Bug report

### Required information:

- Operating system and version:
- Windows 10 Pro Build 19043.1052
- Running a ROS2 Foxy on  a linux Docker  

- OpenRMF installation type:
 I use this tutorial to install RMF: https://osrf.github.io/ros2multirobotbook/intro.html

- ROS distribution and version:
   ROS2 Foxy 
- ROS installation type:
  -Dockerfile

### Description of the bug
I followed the explanation of the tutorial i got no errors.
untill I hat to execute:
 $ ros2 launch demos airport_terminal.launch.xml
 
Then the following error was displayed:
 Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.9', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2launch/command/launch.py", line 159, in main
    return launch_a_launch_file(
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2launch/api/api.py", line 151, in launch_a_launch_file
    launch_service = launch.LaunchService(
TypeError: __init__() got an unexpected keyword argument 'noninteractive'

The same error will be shown independet of which demo rmf  I run.

marcoag, 2021-07-06:
I understand you've got ROS inside a docker container. Did you add the install instructions RMF to the dockerfile? if so it would be useful if you could provide it so we can reproduce the issue.

FerusOlin, 2021-07-06:
I just installed the rmf  in side the docker container through the terminal.

marcoag, 2021-07-06:
I can't seem to reproduce the issue on my side so my recommendation is that you add your steps to the dockerfile and pass it to us for reproducibility (also without it you will lose your changes if the docker shuts down). 

Are you able to create X Windows from your docker? i.e. can you run gazebo? (this might require extra configurations steps).
 
Alternatively you could try this Dockefile that compiles the rmf source code over the ros foxy image: 

<details closed>
<summary>Dockerfile</summary>
<br>

```
FROM ros:foxy

RUN apt-get update \
  && apt-get install -y \
    cmake \
    curl \
    git \
    python3-colcon-common-extensions \
    python3-requests \
    python3-shapely \
    python3-vcstool \
    python3-yaml \
    qt5-default \
    wget \
  && rm -rf /var/lib/apt/lists/*

# setup keys
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys D2486D2DD83DB69272AFE98867170598AF249743

# setup sources.list
RUN . /etc/os-release \
    && echo "deb http://packages.osrfoundation.org/gazebo/$ID-stable `lsb_release -sc` main" > /etc/apt/sources.list.d/gazebo-latest.list

RUN mkdir $HOME/rmf_demos_ws
WORKDIR $HOME/rmf_demos_ws
RUN mkdir src
RUN rosdep update --rosdistro $ROS_DISTRO
RUN wget https://raw.githubusercontent.com/open-rmf/rmf/main/rmf.repos \
      && vcs import src < rmf.repos \
      && apt-get update \
    && apt-get upgrade -y \
    && rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO -y \
    && rm -rf /var/lib/apt/lists/*
RUN apt-get update \
    && apt-get install -y ignition-dome \
    && rm -rf /var/lib/apt/lists/*

RUN . /opt/ros/$ROS_DISTRO/setup.sh \
  && colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

</details>

I'm using [rocker](https://github.com/osrf/rocker) for an easier way to run X applications through my nvidia card but you can directly run docker as long as you're able to run the graphics. Once your image is ready you only need to:

```
rocker --x11 --nvidia rmf:test
# source install/setup.bash
# ros2 launch rmf_demos office.launch.xml
```
 
I recommend trying the smallest demo first (office) and move to the rest once that one works.

Once you have the demo running you will also need network access to access the [rmf-panel](https://github.com/open-rmf/rmf_demos#rmf-panel) through your browser and be able to submit tasks to the system.

marcoag, 2021-07-22:
I will go ahead and close this issue due to inactivity. Feel free to reopen it if there's any persisting `RMF` misbehavior.

closed 2021-07-22 by marcoag