open-rmf/rmf_ros2 #82: Investigate data races reported by Thread Sanitizer #46

aaronchongth, 2021-07-15:
To recreate this build, clone everything on `main` in a new workspace

```bash
export CC=clang
export CXX=clang++

cd ws
source /opt/ros/foxy/setup.bash
colcon build --mixin tsan --cmake-args -DCMAKE_BUILD_TYPE=Debug --packages-up-to rmf_fleet_adapter
```

Package tests can be run individually, for example,

```bash
source install/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

# This one seems the most prominent.
./build/rmf_fleet_adapter/test_rmf_fleet_adapter
```

Quickly skimming through it, I can see that the first topic of interest would be the constructor of `rmf_traffic_ros2::blockade::BlockadeNode`, which similarly uses the node context. The race conditions happen on both `rmw_fastrtps_cpp` and `rmw_cyclonedds_cpp`.

The log can still be downloaded from the `tsan` action in #46 
[4_tsan_build_test.txt](https://github.com/open-rmf/rmf_ros2/files/6822191/4_tsan_build_test.txt)

aaronchongth, 2021-07-29:
Per VC, we should attempt this build on Galactic to see if any of errors are fixed.

aaronchongth, 2021-08-05:
Running the `test_rmf_fleet_adapter` in Galactic, we are getting less data races, but some are still present, https://gist.github.com/aaronchongth/7481bac63c018de0c9bcf71d720b633f

aaronchongth, 2021-08-05:
First data race in `test_TrafficLight`,
* Creation of blockade node, https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/test/adapters/test_TrafficLight.cpp#L170-L172
* Creation of mock adapter, https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/test/adapters/test_TrafficLight.cpp#L200-L202

aaronchongth, 2021-09-02:
This is an effort for https://github.com/open-rmf/rmf_ros2/issues/114.

mxgrey, 2021-10-01:
I've created two PRs that seem to generally eliminate the data races (although it's possible that some rare/improbable data races are still lurking somewhere in the tests):

https://github.com/open-rmf/rmf_traffic/pull/46
https://github.com/open-rmf/rmf_ros2/pull/129

mxgrey, 2021-10-01:
As I continue to run the tests, I do see some data races happening occasionally, but they all seem to originate from the DDS implementations. Those may be difficult for us to do anything about.

closed 2021-11-03 by mxgrey