open-rmf/rmf_ros2 #119: Forfeit Tasks

Yadunund, 2021-09-14:
Introduce a mechanism where a fleet adapter can notify the dispatcher that it can no longer perform a previously assigned task. The dispatcher should then re-initiate the bidding process for this task.

cnboonhan, 2021-09-14:
This seems to be the design for https://github.com/open-rmf/rmf_task/pull/39/files#diff-1d39b8181282eab5320c93dd653c0aa2eb1d5270a3c86cf7a2816ca9cf973547R245

mxgrey, 2021-09-14:
The cancel is a bit different from a forfeit. Cancelling is requested by the operator. A forfeit would be if the active task itself determines that it cannot be completed by the robot.

The redesign will introduce a [`Failed`](https://github.com/open-rmf/rmf_task/blob/724eb0a3bdd60b9e83800c25d8f598e49396be9f/rmf_task/include/rmf_task/Condition.hpp#L77-L80) and a [`Blocked`](https://github.com/open-rmf/rmf_task/blob/724eb0a3bdd60b9e83800c25d8f598e49396be9f/rmf_task/include/rmf_task/Condition.hpp#L60-L62) condition status for task phases. This could be watched to determine when a task cannot be completed.

It's not obvious to me when/how the handling of task failures should be automated. I wonder if maybe the `Request` should include a parameter that says whether the task should be automatically reassigned on failure vs brought to the attention of an operator to decide what to do about the failure.

Yadunund, 2021-09-14:
In addition to task failures/blockages, tasks may need to be forfeited if all the robots in the fleet become unavailable/ get [removed](https://github.com/open-rmf/rmf_ros2/issues/105). 

This intention of this ticket is to brainstorm the topics/messages that will need to be introduced along with implementation changes within the `Dispatcher` node and `FleetUpdateHandle`.