open-rmf/rmf_ros2 #123: Health status for some entities

koonpeng, 2021-09-20:
## Feature request

There should be a health status for at least the following entities

* Doors
* Lifts
* Dispensers
* Ingestors
* Fleets
* Robots

### Implementation considerations

The health statuses should be in a different topic from the state messages so that it is still possible to receive the health status even if the adapters crash for any reason. They should also be published from a node with failover for better resilience.

### Alternatives

### Additional information

mxgrey, 2021-09-20:
Could you elaborate on what health means, if it's something different from state? In DDS there are topic heartbeats that can let you know a node is still up and able to publish, even if it hasn't published anything in a while, but since you're talking about the "health" coming from a different node, I guess you're asking for something else.

We're working towards a point where every node has reliable fail over. If we assume that all state publishers will (eventually) have reliable fail over mechanisms, then is there something else you're looking for that would indicate "health"?

koonpeng, 2021-09-22:
I imagine the meaning of "health" would be different for each type of entities. In general, I would define it loosely as "an entity having an abnormal state which might require user intervention". For example a state publishing node (without fail over) that crashed would be an important event that the user should be aware of.

In most cases, it could be part of the state, but it would be good if there more information is available for the cause of "bad health". e.g. A lift is down because of scheduled maintenance. The case for a separate health node/topic is mostly when it is not possible to have a reliable fail over, e.g. a door/lift adapter that uses embedded hardware.