open-rmf/rmf_ros2 #141: RMF Core Fleet Adapter Loop request bid proposal can't dispatch requests

chadlimjinjie, 2021-11-17:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 20.04.3 LTS
- OpenRMF installation type:
  - Source, Binaries
- OpenRMF version or commit hash
  - Using internal copy of OpenRMF foxy
- ROS distribution and version:
  - 
- ROS installation type:
  - https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html
- Package or library, if applicable:
  - 

### Description of the bug

We have a working demo setup and another setup to integrate open rmf to our robots.

On the working setup when a Loop is send to rmf core, rmf core would create the loop task with BidProposal cost and then select that task as the winning bid which then dispatch loop request.

On our setup when Loop is send to rmf core, rmf core create the loop task with BidProposal cost and then doesn't dispatch the loop requests.

### Steps to reproduce the bug

### Expected behavior

[image]
Our setup should receive a dispatch request according to the winning bid.

### Actual behavior

[image]
We are not receiving a winning bid hence there is no /rmf_task/dispatch_request

### Screenshots

[image]
Bid Notice

[image]
Bid Proposal

### Additional information

mxgrey, 2021-11-17:
The failure to receive a bid generally means that the task request was impossible for any of the fleets in your system to perform.

For a loop task this usually means that you've asked the robot to loop between two waypoints that aren't connected by any path on the navigation graph. Usually when two points on the navigation graph aren't connected, it's because of a small, accidental break in the navigation graph. See this post for an example of what I mean: https://github.com/open-rmf/rmf/discussions/59#discussioncomment-866410 

I recommend closely double-checking your navigation graph in the traffic editor and seeing if there are any lanes whose waypoints are accidentally disconnected.

Yadunund, 2021-11-17:
I noticed that the requested start time for your loop request with real robots (without gazebo) is still `description: {start_time: {sec:0 , nanosec:0}}`. This actually corresponds to the start of unix time (`January 1, 1970`). So you should be populating this time with the present unix time. But this should not have an impact in the dispatcher publishing a `DispatchRequest`. 
There was an issue before related to supporting start_times well in the past but this has been [fixed](https://github.com/open-rmf/rmf_task/pull/22). Could you check if your "internal copy of RMF" as stated in your description includes this fix? Are you using binaries or building from source? Your description states both which is confusing.

More importantly, could you share the entire output of the `BidProposal` message? Specifically highlighting [these fields](https://github.com/open-rmf/rmf_internal_msgs/blob/71e039dbc1056b79543b81dfc5900624f94f25bd/rmf_task_msgs/msg/BidProposal.msg#L11-L21)? Also for a sanity check, is the `use_sim_time` parameter set to `false` for all RMF and fleet adapter nodes that you are running? Also could you try the latest `main` for all the RMF repos and let us know if your problem still persists.

mxgrey, 2021-11-18:
> would the submission time matter for the selection of the winning bid.

I think I understand the issue now. Unfortunately right now there's a [hardcoded cut-off of 2 seconds](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_task_ros2/src/rmf_task_ros2/Dispatcher.cpp#L84-L85) for all bids to arrive. We've generally found that 2 seconds is usually an order of magnitude more time than a bid usually takes to arrive, so we're using this hardcoded timeout until we have a chance to change the implementation to a more sound handshake-based system.

If your task bid takes an unusually long time to generate, you could try extending the 2 second timeout. Or a better stop-gap measure would be to add a ROS parameter for that timeout. This problem will be fixed correctly when we improve the bidding system, but forcing the bid window to be bigger should help in the short term.

mxgrey, 2021-11-18:
It was just pointed out to me that the bidding time is already a ROS parameter. You can set it by modifying [this field](https://github.com/open-rmf/rmf_demos/blob/main/rmf_demos/launch/common.launch.xml#L11v) in the `common.launch.xml`.

chadlimjinjie, 2021-11-18:
> I noticed that the requested start time for your loop request with real robots (without gazebo) is still `description: {start_time: {sec:0 , nanosec:0}}`. This actually corresponds to the start of [...]

I'll have to check with my team on what version we are using. But currently I am using the 'feature/integration-test' branch the winning bid is still not selected and no DispatchRequest is published. 

> Are you using binaries or building from source? Your description states both which is confusing.

I'm currently using https://github.com/open-rmf/rmf 'feature/integration-test' branch build from source. 

> More importantly, could you share the entire output of the `BidProposal` message? Specifically highlighting [these [...]

Here are the logs from the BidProposal message.

[image]
without Gazebo

[image]
without Gazebo `use_sim_time` parameter set to `false`.

cnboonhan, 2021-11-18:
hello,

is there a specific reason for choosing to use integration-tests branch? I made the branch as a legacy "bookmark"  to allow some previous tooling to work, but I would strongly recommend working off something more recent if possible.

Would also recommend uninstalling any / all rmf binaries if you are also building RMF from source to prevent overlay confusion:
```
sudo apt remove ros-$ROS_DISTRO-rmf-*
```
should do the trick.

KaeYan, 2021-11-19:
> So you should be populating this time with the present unix time. But this should not have an impact in the dispatcher publishing a `DispatchRequest`. 

@Yadunund if i set the `use_sim_time` to `false`, how can we config the fleet adapter to use unix time?

Also, when we send fleet states, we want to use unix time. Say if the unix time now in nanoseconds is 1637302549019356595. How should we update the `Time` in `Location`? Is it secs = 1637302549, then nanoseconds = 1637302549019356595?