open-rmf/rmf_ros2 #148: linking error when building rmf_fleet_adapter

mayman99, 2021-12-20:
## Bug report
`
defined reference to symbol 'pthread_rwlock_unlock@@GLIBC_2.2.5' make
`
### Required information:

- Operating system and version:
  - Ubuntu 20.04
- OpenRMF installation type:
  - source
- OpenRMF version or commit hash
  - latest, 66cd1dc
- ROS distribution and version:
  - 
- ROS installation type:
  - binary
- Package or library, if applicable:
  - rmf_fleet_adapter

### Description of the bug

fixed by adding `-pthread` to `experimental_lift_watchdog` target link libs
```
target_link_libraries(experimental_lift_watchdog
  PUBLIC
    -pthread
  PRIVATE
    ${rclcpp_LIBRARIES}
    ${rmf_fleet_msgs_LIBRARIES}
    ${std_msgs_LIBRARIES}
)
```

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

mxgrey, 2021-12-22:
Linkers are awfully mysterious.. I have no idea how you're the first one to be dealt this error.

There's a preferable fix to this problem than explicitly linking `libpthread`:

```
find_package(Threads REQUIRED)
...
target_link_libraries(target_name [PUBLIC|PRIVATE] Threads::Threads)
```

That should fix the linking problem while also being cross-platform.

Yadunund, 2022-05-12:
Potentially fixed by https://github.com/open-rmf/rmf_ros2/pull/204

closed 2023-07-12 by mxgrey