open-rmf/rmf_ros2 #149: Question: fleet adapter not accepting ChargeBattery task

chadlimjinjie, 2021-12-22:
Hi just want to check how to populate Task for a charging job right now I have populated the task_type TYPE_CHARGE_BATTERY=3 as well as the Station field to the charging station. However when I  submit this task to rmf, rmf says that my fleet is not configured to accept ChargeBattery task. 

What am I missing/got wrong here? Anything I'll have to configure in the ros parameter for fleet adapter?

[image]

mxgrey, 2021-12-22:
The current implementation of the fleet adapters takes care of charging automatically. There isn't currently a way to explicitly demand that a robot perform a charging task.

We're working on a redesign of the task system. We can take this into account so that the new system supports explicit charging tasks, although I can't think of a reason that a user would need to explicitly ask for a charging since the task planner will make optimal decisions about when charging should occur.

Maybe what you really want is to set a [finishing request](https://github.com/open-rmf/rmf_ros2/blob/a88fd242009ca1f92b4ccc4f61940053f4440905/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/FleetUpdateHandle.hpp#L111-L113) which decides what the behavior of the robot will be once it has finished all of its assigned tasks. You can set the finishing request to [charge the battery](https://github.com/open-rmf/rmf_task/blob/a548bacb2b5ede18a7cb5a9e40322a1b808d39be/rmf_task/include/rmf_task/requests/ChargeBatteryFactory.hpp#L41-L42) so that the robot automatically returns to it charger every time it is finished with its task queue.

chadlimjinjie, 2021-12-22:
I see okay I'll look for a work around for what we need right now. Cause what we need is to pass rmf waypoint name to our system so that our system uses the rmf waypoint name as the charging station.

closed 2021-12-22 by chadlimjinjie

mxgrey, 2021-12-22:
If you can describe what kinds of information/messages from RMF would be helpful for your system, let us know and we can try to incorporate it into the battery charging workflow. E.g. Should the battery charging task publish a message like
```
robot_name: "MyRobot001"
charging_station: "charger_004"
```

when RMF wants `"MyRobot001"` to use `"charger_004"` to charge?

chadlimjinjie, 2021-12-22:
> If you can describe what kinds of information/messages from RMF would be helpful for your system, let us know and we can try to incorporate it into the battery charging workflow. E.g. Should the [...]

Yes that looks sufficient to what we need and also the robot needs to know how to get there through PathRequest messages from rmf to our fleet driver

chadlimjinjie, 2021-12-22:
Just one more question if you guys happen to see this. We also have an issue that rmf will publish the PathRequest to the closes path first and so happens our ending point is closes the the first waypoint and once our robot reaches the first waypoint it returns to our ending point next.

mxgrey, 2021-12-22:
> Just one more question if you guys happen to see this. We also have an issue that rmf will publish the PathRequest to the closes path first and so happens our ending point is closes the the first waypoint and once our robot reaches the first waypoint it returns to our ending point next.

This is a byproduct of using the `rmf_fleet_msgs` API which has some known issues like what you're describing. Unfortunately there isn't a very good way to fix this behavior because the `rmf_fleet_msgs` API doesn't communicate enough information to the fleet adapter.

You can consider doing a more direct integration with the fleet adapter API to prevent this kind of issue. [This fleet adapter template](https://github.com/open-rmf/fleet_adapter_template) can help provide a baseline for integrating your fleet.