open-rmf/rmf_ros2 #157: build error about "rmf_cmake_uncrustify" in rmf_traffic_ros2

dujeong, 2022-01-24:
## Bug report

### Required information:

- Operating system and version: 
  - Ubuntu 20.04 (docker)
- OpenRMF installation type: 
  - From source
- OpenRMF version or commit hash
  - 
- ROS distribution and version:
  - foxy
- ROS installation type:
  - from binaries
- Package or library, if applicable:
  - 

### Description of the bug

when I build the source code by cmdline("colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release"), below error occured.
```
Starting >>> rmf_traffic_ros2
Starting >>> rmf_visualization_fleet_states
Finished <<< rmf_visualization_fleet_states [5.47s]                                                                         
--- stderr: rmf_traffic_ros2                                                                                                
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "Findrmf_cmake_uncrustify.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "rmf_cmake_uncrustify", but CMake did not find one.

  Could not find a package configuration file provided by
  "rmf_cmake_uncrustify" with any of the following names:

    rmf_cmake_uncrustifyConfig.cmake
    rmf_cmake_uncrustify-config.cmake

  Add the installation prefix of "rmf_cmake_uncrustify" to CMAKE_PREFIX_PATH
  or set "rmf_cmake_uncrustify_DIR" to a directory containing one of the
  above files.  If "rmf_cmake_uncrustify" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< rmf_traffic_ros2 [5.60s, exited with code 1]
Aborted  <<< rmf_traffic_editor_test_maps [16.6s]                                                                                      
Aborted  <<< rmf_demos_maps [1min 7s]                                                                                                      
Aborted  <<< rmf_robot_sim_gazebo_plugins [1min 25s]

...
```

### Steps to reproduce the bug

### Expected behavior

### Actual behavior

### Screenshots

### Additional information

Yadunund, 2022-01-24:
Hi @dujeong ,

Could you make sure to pull in the latest `main`  for `rmf_ros2` and then perform a clean build which includes running `rm -rf build/ install/ log/` inside your workspace? As of this PR https://github.com/open-rmf/rmf_ros2/pull/140, we updated the `CMakeLists.txt` to use the built-in `ament_cmake_uncrustify` instead of `rmf_cmake_uncrustify`.

dujeong, 2022-01-24:
thanks for your reply, @Yadunund 

Do I have to change other packages all?

Below is the repositories I used.
```
repositories:
  rmf/rmf_battery:
    type: git
    url: https://github.com/open-rmf/rmf_battery.git
    version: foxy
  rmf/rmf_internal_msgs:
    type: git
    url: https://github.com/open-rmf/rmf_internal_msgs.git
    version: foxy
  rmf/rmf_ros2:
    type: git
    url: https://github.com/open-rmf/rmf_ros2.git
    version: foxy
  rmf/rmf_task:
    type: git
    url: https://github.com/open-rmf/rmf_task.git
    version: foxy
  rmf/rmf_traffic:
    type: git
    url: https://github.com/open-rmf/rmf_traffic.git
    version: foxy
  rmf/rmf_utils:
    type: git
    url: https://github.com/open-rmf/rmf_utils.git
    version: foxy
  rmf/ament_cmake_catch2:
    type: git
    url: https://github.com/open-rmf/ament_cmake_catch2.git
    version: foxy
  rmf/rmf_visualization:
    type: git
    url: https://github.com/open-rmf/rmf_visualization.git
    version: foxy
  rmf/rmf_visualization_msgs:
    type: git
    url: https://github.com/open-rmf/rmf_visualization_msgs.git
    version: foxy
  rmf/rmf_building_map_msgs:
    type: git
    url: https://github.com/open-rmf/rmf_building_map_msgs.git
    version: foxy
  rmf/rmf_simulation:
    type: git
    url: https://github.com/open-rmf/rmf_simulation.git
    version: foxy
  rmf/rmf_traffic_editor:
    type: git
    url: https://github.com/open-rmf/rmf_traffic_editor.git
    version: foxy
  demonstrations/rmf_demos:
    type: git
    url: https://github.com/open-rmf/rmf_demos.git
    version: foxy
  thirdparty/menge_vendor:
    type: git
    url: https://github.com/open-rmf/menge_vendor.git
    version: foxy
```

Yadunund, 2022-01-24:
Ah from your original description, I assumed you were building latest `main` from source. If you want to build the `foxy` version of the packages, the quickest solution would be to also clone in this package into your workspace https://github.com/open-rmf/rmf_cmake_uncrustify

But to fix the problem in the long run, we could either
1) Backport the `CMakeLists.txt` changes to the `foxy` branches for the various repos
2) Maintain a `foxy` branch on https://github.com/open-rmf/rmf where the `rmf.repos` file still includes `rmf_cmake_uncrustify`

dujeong, 2022-01-24:
Whenever error occurs, I delete the folder(build/, install/, log/) and build again.
And then I succeeded!
But there are many dependencies to install..
(Ceres, python3-requests, ros-foxy-pybind11-vendor, ...)

Anyway, Thanks a lot!

closed 2022-01-24 by dujeong

Yadunund, 2022-02-03:
> But there are many dependencies to install..
(Ceres, python3-requests, ros-foxy-pybind11-vendor, ...)

Those dependencies should get installed when you follow the `Additional Dependencies` and run `rsodep install...` (make sure you replace `--rosdistro galactic` with  `--rosdistro foxy` if you're using `foxy`) instructions in the [README](https://github.com/open-rmf/rmf#additional-dependencies)