open-rmf/rmf_ros2 #164: redesignv2: time fields are not populated

cnboonhan, 2022-02-03:
It seems some fields, such as `unix_millis_start_time` and `unix_millis_finish_time` are populated as `null` . This might give issues when trying to visualise task durations on the dashboard.

Reproducing steps
```
ros2 launch rmf_demos_ign office.launch.xml headless:=0 server_uri:=ws://localhost:8001 use_sim_time:=true
rmf_api_server
ros2 run rmf_demos_tasks dispatch_loop -s pantry -f hardware_2 -n 3 --use_sim_time

firefox localhost:8000/docs
GET /tasks with task_id patrol.dispatch.0
```
Results
```
[
  {
    "booking": {
      "id": "patrol.dispatch-0",
      "unix_millis_earliest_start_time": 37570,
      "priority": null,
      "labels": null
    },
    "category": "Patrol",
    "detail": {},
    "unix_millis_start_time": null,
    "unix_millis_finish_time": null,
    "original_estimate_millis": 246578,
    "estimate_millis": 270720,
    "assigned_to": {
      "group": "tinyRobot",
      "name": "tinyRobot2"
    },
    "status": "underway",
    "dispatch": null,
    "phases": {
      "1": {
        "id": 1,
        "category": "Go to [place:pantry]",
        "detail": "Moving the robot from [place:tinyRobot2_charger] to [place:pantry]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": 27344,
        "estimate_millis": 24142,
        "final_event_id": 0,
        "events": {
          "0": {
            "id": 0,
            "status": "underway",
            "name": "Go to [place:pantry]",
            "detail": "Moving the robot from [place:tinyRobot2_charger] to [place:pantry]",
            "deps": [
              1
            ]
          },
          "1": {
            "id": 1,
            "status": "underway",
            "name": "Move [tinyRobot/tinyRobot2] to ( 16.8463 -5.40407  1.57722)",
            "detail": {},
            "deps": []
          }
        },
        "skip_requests": null
      },
      "2": {
        "id": 2,
        "category": "Go to [place:hardware_2]",
        "detail": "Moving the robot from [place:pantry] to [place:hardware_2]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": null,
        "estimate_millis": 49324,
        "final_event_id": null,
        "events": null,
        "skip_requests": null
      },
      "3": {
        "id": 3,
        "category": "Go to [place:pantry]",
        "detail": "Moving the robot from [place:hardware_2] to [place:pantry]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": null,
        "estimate_millis": 49302,
        "final_event_id": null,
        "events": null,
        "skip_requests": null
      },
      "4": {
        "id": 4,
        "category": "Go to [place:hardware_2]",
        "detail": "Moving the robot from [place:pantry] to [place:hardware_2]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": null,
        "estimate_millis": 49324,
        "final_event_id": null,
        "events": null,
        "skip_requests": null
      },
      "5": {
        "id": 5,
        "category": "Go to [place:pantry]",
        "detail": "Moving the robot from [place:hardware_2] to [place:pantry]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": null,
        "estimate_millis": 49302,
        "final_event_id": null,
        "events": null,
        "skip_requests": null
      },
      "6": {
        "id": 6,
        "category": "Go to [place:hardware_2]",
        "detail": "Moving the robot from [place:pantry] to [place:hardware_2]",
        "unix_millis_start_time": null,
        "unix_millis_finish_time": null,
        "original_estimate_millis": null,
        "estimate_millis": 49324,
        "final_event_id": null,
        "events": null,
        "skip_requests": null
      }
    },
    "completed": [],
    "active": 1,
    "pending": [
      2,
      3,
      4,
      5,
      6
    ],
    "interruptions": null,
    "cancellation": null,
    "killed": null
  }
]
```

mxgrey, 2022-02-03:
Thanks for the detailed report!

This should now be fixed on `redesign_v2` by this commit: 50edb27788937fc9c2a52e82de664794766a9f75

Feel free to reopen if you notice any remaining issues with this.

closed 2022-02-03 by mxgrey

youliangtan, 2022-02-09:
Pertaining to this, on top of timing info for active task, also populated timing info for tasks that are queued: https://github.com/open-rmf/rmf_ros2/commit/97e5a0cdb975a2407f2a885d5ec85d87498fc76
[image]