open-rmf/rmf_ros2 #169: RMF with 2 linked buildings

HappySamuel, 2022-02-09:
Hi

Is there a way that RMF can be loaded with 2 buildings .yaml files? As the robot is requested to travel between 2 buildings (which consist of L2 of 2 buildings linked with a bridge. Any suggestions / guidance on this??

Best,
Samuel

codebot, 2022-02-09:
Hi Samuel,

That sounds very interesting! I'd love to see the buildings!

There are several ways to do this. At the moment, fleet adapters use a single graph for their traffic networks, so this comes down to a problem of creating a large, "multi-building" traffic graph, if I understand the question correctly.

One method is to use graphics-editing tools like `GIMP` to make "composite floorplan" images that combine each building's corresponding levels into single (huge) floorplan images. For example, a single large image which combines both buildings' L1, and another large image that combines both buildings' L2 and the bridge, etc. However, if the buildings are very far apart, the composite images might be too huge to be practical.

Another method is to use some very fresh features recently merged into the https://github.com/open-rmf/rmf_traffic_editor `rmf_traffic_editor` repo, which allow maps to be described in latitude / longitude units, rather than in "pixels of the reference floorplan image". If you are building the `rmf_traffic_editor` repo from source, you'll have this functionality available.

When you create a new building map, the `traffic-editor` GUI will ask if you want to use a geo-referenced coordinate system. If you create a map in geo-referenced mode, coordinates are stored in `.building.yaml` files in latitude/longitude, and traffic-editor will stream OpenStreetMap tiles as you zoom and pan around, so that you can position the floorplan images in global (lat/lon) coordinates, using the "Layers" feature of traffic-editor. [This process does work, although it's not a particularly great UX at the moment, since it involves a lot of trial-and-error when hand-editing transformation parameters to position the layers reasonably.] You'll also need to specify a coordinate system for `rmf_building_map_tools` to use when projecting latitude/longitude into a Cartesian navigation graph for use by the RMF planner. Each region of the world has its own favorite way to do this, with a corresponding EPSG code (you can use http://epsg.io/ to find one relevant for your region). This is brand new, not-yet-documented stuff in `traffic-editor` and `rmf_building_map_tools`, so you may find some issues/bugs with this workflow. But if your campus is very large, this might be more practical than the "composite floorplan images" approach.

Note: at time of writing, the map-tile server for `traffic-editor` is only loaded with detailed data for Singapore and Malaysia. We are working on bringing up a "whole-planet" tile server now. It's a lot of data :smile: 

Cheers!
-Morgan

HappySamuel, 2022-02-11:
Hi Morgan

There're several buildings in Singapore that are inter-linked with a bridge at L2, such as newly launched NorthShorePlaza 1&2 (hasn't updated in Google Map) and also inter-linked buildings that next to Jurong East MRT area. It seems like the inter-linked buildings are the trend in Mall construction.

Yes, you understand the question correctly. 

Thanks for the suggested methods, will try out above methods and see which one suits the best.

Another question, are there any 2D map_server / 3D maps localization -like packages that support latitude/longitude defining? that can be used for the 2nd method you suggested. 

This is because we're currently using 2D map for navigation while 3D map for localizing, but merging of 3D maps create issue. We have been thinking of if this concept is correct for deploying? Having 2 buildings linked in RMF as below image. 
[image]

Best,
Samuel

HappySamuel, 2022-04-01:
Hi @codebot 

Thanks for the suggested method (1), i am able to try it out on gazebo simulation successfully. Here i attached the video link of how it run with RMF.
[Interlink buildings with RMF on Gazebo - 8x speedup video](https://drive.google.com/file/d/14KWb1XqfyXY5JveHCKAg-HNEbbV1IvrV/view?usp=sharing)

Best,
Samuel

closed 2022-04-01 by HappySamuel