open-rmf/rmf_ros2 #177: Fullcontrol path requests has waypoints with wrong yaw values

idkwhatmyid, 2022-02-23:
When x coordinates of consecutive waypoints are decreasing the yaw value is always 0.0 but it needs to face the other direction (3.14). The same goes for decreasing y coordinates where it is 1.57 but needs to be -1.57. I also notice robot spins to 0.0 degrees infront of the doors. What node is providing the yaw values in the `robot_path_requests` messages?

```
path:
- t:
    sec: 1851
    nanosec: 745000000
  x: 48.25251388549805
  y: -19.186559677124023
  yaw: 1.5216608047485352
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
- t:
    sec: 1852
    nanosec: 42178250
  x: 48.25251388549805
  y: -19.186559677124023
  yaw: 1.5658183097839355
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
- t:
    sec: 1856
    nanosec: 579068139
  x: 48.242881774902344
  y: -21.121646881103516
  yaw: 1.5658183097839355
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
- t:
    sec: 1859
    nanosec: 465962349
  x: 48.242881774902344
  y: -21.121646881103516
  yaw: 0.013681734912097454
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
- t:
    sec: 1869
    nanosec: 817861630
  x: 43.40037155151367
  y: -21.15367317199707
  yaw: 0.013681734912097454
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
- t:
    sec: 1870
    nanosec: 62518684
  x: 43.40037155151367
  y: -21.15367317199707
  yaw: -0.016246803104877472
  obey_approach_speed_limit: false
  approach_speed_limit: 0.0
  level_name: L1
  index: 0
task_id: '1473'
```
The robot appears to  spin randomly at turning points and in front of doors. We are using ROS2 free_fleet and Nav2. Can someone help to fix this behaviour? Thanks.

mxgrey, 2022-02-23:
If your robot is designed to generally not drive in reverse, then be sure to set the [`reversible`](https://github.com/open-rmf/rmf_ros2/blob/14ba7bbfc6da10bcde1e00ea5e4371332525fccc/rmf_fleet_adapter/launch/fleet_adapter.launch.xml#L23) parameter to `false`. By default that parameter is `true`, so the robot will prefer to drive in reverse if the alternative would be to spend time turning 180-degrees.

The default was set to `true` because our earliest testing platforms for RMF were agnostic towards driving forwards or backwards, but I've been thinking about changing that default to `false` considering how many robots (especially commercial robots) seem to only want to drive forwards :sweat_smile:.

idkwhatmyid, 2022-02-25:
Thank you, changing the parameter fixed it.

closed 2022-02-25 by idkwhatmyid