open-rmf/rmf_ros2 #185: Incorrect Reporting of event status when robot is sent to waypoint it already is at.

kerliang0801, 2022-03-14:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 20.04
- OpenRMF installation type:
  - From source
- OpenRMF version or commit hash
dbaae57b6a451236408843534ebb1ebd3805b96f
- ROS distribution and version:
Galactic
- ROS installation type:
Binaries

### Description of the bug
Incorrect reporting when you ask a robot to move to whatever node it already is at. Status will always be "underway" via the GET `/tasks` API.

### Steps to reproduce the bug
1. launch rmf_api_server
2. In a terminal, `ros2 launch rmf_demos_gz hotel.launch.xml server_uri:=ws://localhost:8001/ headless:=0 use_sim_time:=true`
3. Send a robot task via the tasks/robot_task endpoint (I'm not too sure if the json below is supposed to look like this, seems a bit funky)
```
{
    "type": "robot_task_request",
    "robot": "deliveryBot_1",
    "fleet": "deliveryBot",
    "request": {
        "category": "compose",
        "description": {
            "category": "compose",
            "phases": [
                {
                    "activity": {
                        "category": "sequence",
                        "description": {
                            "activities": [
                                {
                                    "category": "go_to_place",
                                    "description":  "deliverybot_charger"
                                }
                                ,{
                                    "category": "perform_action",
                                    "description": {
                                        "unix_millis_action_duration_estimate": 0,
                                        "category": "teleop",
                                        "description": {}
                                    }
                                }
                            ]
                        }
                    }
                }
            ]
        }
    }
}
```
4. Call GET /tasks. This is the response i get. Note that status under events will always be underway for the Go to place event even when completed is 1.
```
 {
    "booking": {
      "id": "c6a00b4e-3d3e-4fc6-b401-d0aae963c5a9",
      "unix_millis_earliest_start_time": 1504206,
      "priority": null,
      "labels": null
    },
    "category": "compose",
    "detail": {},
    "unix_millis_start_time": 1504605,
    "unix_millis_finish_time": 2198249,
    "original_estimate_millis": 0,
    "estimate_millis": 0,
    "assigned_to": {
      "group": "deliveryBot",
      "name": "deliveryBot_1"
    },
    "status": "underway",
    "dispatch": null,
    "phases": {
      "1": {
        "id": 1,
        "category": "Sequence",
        "detail": "[{\"category\":\"Go to [place:deliverybot_charger]\",\"detail\":\"Moving the robot from [place:deliverybot_charger] to [place:deliverybot_charger]\"},{\"category\":\"Perform action\",\"detail\":\"Performing action teleop at waypoint [[place:deliverybot_charger]]\"}]",
        "unix_millis_start_time": 1504605,
        "unix_millis_finish_time": null,
        "original_estimate_millis": 0,
        "estimate_millis": 0,
        "final_event_id": 0,
        "events": {
          "0": {
            "id": 0,
            "status": "underway",
            "name": "Sequence",
            "detail": {},
            "deps": [
              1,
              2
            ]
          },
          "1": {
            "id": 1,
            "status": "underway" <--!!,
            "name": "Go to [place:deliverybot_charger]",
            "detail": "Moving the robot from [place:deliverybot_charger] to [place:deliverybot_charger]",
            "deps": []
          },
          "2": {
            "id": 2,
            "status": "standby",
            "name": "Perform action",
            "detail": "Performing action teleop at waypoint [[place:deliverybot_charger]]",
            "deps": []
          }
        },
        "skip_requests": null
      }
    },
    "completed": [1],
    "active": 1,
    "pending": [],
    "interruptions": null,
    "cancellation": null,
    "killed": null
  }
```

Please do let me know if the JSON payload im posting to the `tasks/robot_task` endpoint is incorrect.

mxgrey, 2022-03-15:
This should be fixed by [032c01e](https://github.com/open-rmf/rmf_ros2/pull/186/commits/032c01ebeb304760025885ff6f6442c328bf4af4). To save some hastle, I'm lumping that commit into this broader PR: https://github.com/open-rmf/rmf_ros2/pull/186

mxgrey, 2022-03-17:
Fixed by  #186

closed 2022-03-17 by mxgrey