open-rmf/rmf_ros2 #192: rmf_traffic_schedule keeps dying

royito55, 2022-04-07:
Hi, I'm trying to implement RMF to my own ROS2 simulation. 

I am using as a template the `office_launch.xml` from `rmf_demos_gz`, and I've removed the simulation part because the one I have is already working with the `slotcar` plugin.

I am isolating the `common.launch.xml` from the office launch, but I keep getting the following error:
`[ERROR] [rmf_traffic_schedule-1]: process has died [pid 1591262, exit code 127, cmd '/opt/ros/galactic/lib/rmf_traffic_ros2/rmf_traffic_schedule --ros-args -r __node:=rmf_traffic_schedule_primary --params-file /tmp/launch_params_8s6mnztv'].`

Do I need the other two launches (adapter and robot_state_aggregator) for it to work? I can see that it doesn't take extra parameters so I'm stuck on where the error could be.

The integration documentation is very complicated to follow, so please let me know if there is an example of an adaptation of RMF into a simulation, as I haven't found any

mxgrey, 2022-04-07:
The most common cause of this kind of spurious immediate crash is that you have nodes that were compiled with outdated ROS2 message definitions. They're transmitting messages that are broken from the point of view of the schedule node and causing rclcpp to crash.

You'll need to double-check the hygiene of your RMF build environment. If you're building RMF from source in a colcon workspace, I recommend deleting the `build` and `install` directories and rebuilding from scratch.

If you've installed a binary distro of RMF like `ros-[foxy|galactic]-rmf-*` then I recommend using `sudo apt-get remove ros-<distro>-rmf-*` to get rid of those before rebuilding. The message definitions of RMF that are tied to those ROS2 distros are not compatible with the latest version of RMF and are likely to cause this kind of crashing.

royito55, 2022-04-07:
Thank you, I removed the binary install and I've been trying to build from source following the instructions. But `rmf_traffic_ros2` is not compiling due to a `libyaml-cpp` issue, which I have no idea about. (https://github.com/open-rmf/rmf/issues/146). Have you seen that error before?

Yadunund, 2022-05-12:
Closing since the issue seems to be resolved.

closed 2022-05-12 by Yadunund