open-rmf/rmf_ros2 #224: Task recovery when fleet adapter get restarted

youliangtan, 2022-08-19:
Currently, once a task is assigned to the robot, it will get "stored" in a fleet adapter internal task queue. For whatever reason that triggers a fleet adapter restart, the assigned tasks will get lost after the restart. Therefore, task recovery capability is quite crucial in a RMF system. (for example, user will often update the fleet adapter pod to a newer release, which requires a restart).

There's a pending PR which tries to implement the task backup feature: https://github.com/open-rmf/rmf_ros2/pull/161
In this pr, the task logs and backup are stored in a form of a db, in this case, `.sqlite` format. 

Here, An extended proposal is to leverage on the existing `rmf-web` system. Since we already update the `rmf-web` persistence db 
via WebSocket (with `server_uri` param), the fleet adapter can also query the same database during restart. This results in a simpler architecture. Furthermore, similar to the usage of [Tortoise ORM](https://tortoise-orm.readthedocs.io/en/latest/) in the `api-server`, the fleet adapter should also use an ORM to support various db. (eg. postgresql, sqlite...)

```mermaid
flowchart LR
    A[Fleet Adapter] -->|ws broadcast states| B(rmf-web: api server)
    B <-->|orm| D[database]
    A <-->|orm| D[database]
```

sripadsirik, 2026-10-31:
Hi, I came across this while researching what happens to assigned tasks when a fleet adapter restarts. Has anything changed in how adapters handle this since you filed it, or is re-issuing tasks after a restart still on the operator? Curious how teams running RMF in production treat it today.

aaronchongth, 2026-11-02:
Unfortunately, the fallback at the moment is still upon the operator's side, we have not gotten to working out this feature yet. In production, we generally avoid fleet adapter restarts at all cost, and would advise folks to use the decommissioning feature instead. Any ongoing tasks will not be affected when decommissioning a robot, however assigned but not yet ongoing tasks may be re-assigned to other robots.