open-rmf/rmf_ros2 #230: Checking rmf_task_status dynamically.

Vishal24rg, 2022-09-09:
Hi Team,
we were trying to check the rmf task status for dynamic obstacles.
we are getting the [slotcar_tinyRobot2]: Stopping [tinyRobot2] to avoid a collision but need to check the overall rmf_task_status.
with updated rmf repository were in rmf_ros2 module web-socket package is added.
before we used to see the task status using socket I/O but now not able to see rmf_task status.
How we can see the rmf_task status which topic we need to check rmf_task_status for ex "running", "paused", "stopped", "delayed" ?
[image]

Yadunund, 2022-09-11:
Navigating around dynamic obstacles is the responsibility of the navigation stack as highlighted in a relevant discussion here https://github.com/open-rmf/rmf/discussions/166. The robots in our simulation do not use such a stack to navigate.

All robot and task state information now gets sent over websocket. See this post for more details https://github.com/open-rmf/rmf/discussions/189

If your navigation stack reports an error, you can propagate this error via the RobotState msg sent out via websocket. To do so, your fleet adapter will need to [issue a ticket](https://github.com/open-rmf/rmf_ros2/blob/617e83d32f028f2115885dd4f4f6b66799762db6/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/RobotUpdateHandle.hpp#L292-L306) when this happens. For details, see this previous post https://github.com/open-rmf/rmf/discussions/203#discussioncomment-3566503

Yadunund, 2022-09-11:
If you have any further questions, feel free to open a new ticket on our discussions board https://github.com/open-rmf/rmf/discussions

closed 2022-09-11 by Yadunund