open-rmf/rmf_ros2 #248: Update implementation of FleetUpdateHandle::close_lanes() to automatically replan for robots if needed

Yadunund, 2022-11-10:
The [FleetUpdateHandle::close_lanes()](https://github.com/open-rmf/rmf_ros2/blob/38cbdadae21496767311ee27434213402641ae65/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/FleetUpdateHandle.hpp#L203-L204) allows for lanes in the navigation graph of the fleet to be closed. 

However the [implementation](https://github.com/open-rmf/rmf_ros2/blob/6193007acdb9c7290f1fcb587beb84457ccd6c32/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/FleetUpdateHandle.cpp#L1512-L1556) does not currently replan the itinerary of a robot in the fleet that may be affected by a closure. Currently the fleet adapters have to implement [logic](https://github.com/open-rmf/rmf_ros2/blob/6193007acdb9c7290f1fcb587beb84457ccd6c32/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/FleetUpdateHandle.cpp#L1512-L1556) to decide whether to replan. We can save a lot of repeated code in fleet adapter implementations if this logic was moved upstream.

@mxgrey do you see any issues with moving the replanning logic directly into the`FleetUpdateHandle::close_lanes()` implementation?

mxgrey, 2022-11-10:
I think it's a tricky situation because I don't think simply the fact that a lane is closing is enough information to tell us whether a robot which is moving down that lane _should_ replan and potentially backtrack.

For example, if the lane is being closed because an obstacle has appeared in the middle of the lane, then a robot which is on that lane _but past the obstacle_ should **not** replan.

For simplifying high-level decisions like this, I might suggest that we can put the replanning logic into the `EasyFullControl` with a lane-closing API that requires users to specify an obstacle location so that the `EasyFullControl` adapter knows how to make the right decision.

Yadunund, 2022-11-10:
That sounds good 👌🏼