open-rmf/rmf_ros2 #249: Cannot analyze task_state json message as before

allstarry, 2022-11-19:
Hi Team,
We integrated rmf with our web server a few months ago, we were trying to listen on port 8000 to get task_state json message, but after we switched to the new humble version we found that we cannot analyze task_state json as before due to some schema is empty(like details). I am wondering whether it is a bug or just an update.

Briancbn, 2022-11-23:
If you are using the `simple_api_server` on Humble, please remember to upgrade a few python dependencies, due to bad Python packages releases on debian.

```
python3 -m pip install -U websockets fastapi
```

allstarry, 2022-11-24:
Thanks for your help. We always build from sources on main branch, and I update python dependencies follow your instructions but it does not solve the problem.
After launch RMF I send a task request with command line 
`ros2 run rmf_demos_tasks dispatch_delivery -p p1 -ph coke_dispenser -d p2 -dh coke_ingestor -pp coke,1 -dp coke,1`
Then I catch json messages with wireshark, it shows
` [truncated]{"data":{"active":1,"assigned_to":{"group":"AMR","name":"amr07"},"booking":{"id":"delivery.dispatch--2147483648","unix_millis_earliest_start_time":1669272615286},"category":"Delivery","completed":[],"detail":"","estimate_milli`
As you can see, the detail is still empty.

mxgrey, 2022-11-24:
Can you be more specific about what information you're expecting the take state message to contain based on previous versions and how your expectation is being violated in the newer version?

I don't recall any intentional changes to the task state information that we're publishing in the latest version.

The `detail` field specifically is only meant to be a human-readable hint to describe the task and doesn't have strict requirements on how it gets filled in. That being said, if it used to be filled with something useful but is now blank then that sounds like an undesirable regression somewhere in the code, but we'd need more information on the problem to identify where the cause might be.