open-rmf/rmf_ros2 #256: help ：Sending msg in event causes process has died

liugehaizaixue, 2023-01-04:
Hello, I was looking at the task system recently.
 I have customized an event to implement the process of sending requests to robots and returning status and results, similar to 'DispenseItem'. But 'DispenseItem' is a phase, which I implement in Active following its writing method_ init_ obs().
 The implementation code is shown in the first figure. I use custom ` rmf_waitformessage_msgs ` replaced the original ` rmf_dispenser_msg`. The compilation is successful, but when I send the compose task to call this' event ',' process has died 'appears. I would like to ask you where the mistake occurred.Thank you!
[image]
[image]

mxgrey, 2023-01-19:
I can't necessarily know what's causing the crash without debugging it myself.

To debug this type of crash I recommend these steps:
* Rebuild the colcon workspace with the addition of `--cmake-args DCMAKE_BUILD_TYPE=RelWithDebInfo`
* Run your fleet adapter in gdb until it crashes

That being said, I'm most suspicious of `node`, specifically the way it's being captured by reference in the lambda. The lambda will outlive the reference and that reference will get triggered after the reference has died. You should instead capture `node` by copy, which means deleting the `&` in the capture list.

mxgrey, 2023-01-19:
Since this isn't a bug in RMF or a feature request, I'll be closing this issue. If you want to discuss the matter further I recommend [opening a discussion] (https://github.com/open-rmf/rmf/discussions).

closed 2023-01-19 by mxgrey