open-rmf/rmf_ros2 #259: Fleet adapter crash with no error when doing a multi floor planning

koonpeng, 2023-01-17:
## Bug report

### Required information:

- Operating system and version:
  - ubuntu22.04
- OpenRMF installation type:
  - source
- OpenRMF version or commit hash
  - main
- ROS distribution and version:
  - humble
- ROS installation type:
  - binary
- Package or library, if applicable:
  - rmf_fleet_adapter

### Description of the bug
When a patrol task is sent with waypoints in different floors, the fleet adapter exits silently with no error. The last message it prints is from https://github.com/open-rmf/rmf_ros2/blob/3c17991d6cd003a75986676cc7c0134b8212138b/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/FleetUpdateHandle.cpp#L292-L300.

```
[fleet_adapter-16] [INFO] [1673946622.213851799] [tinyRobot_fleet_adapter]: Planning for [2] robot(s) and [1] request(s)
[ERROR] [fleet_adapter-16]: process has died [pid 1083490, exit code -11, cmd '/home/koonpeng/ws-osrf/cgh/install/lib/rmf_demos_fleet_adapter/fleet_adapter -c office/tinyRobot_config.yaml -n office/nav_graphs/0.yaml --ros-args --params-file /tmp/launch_params_3bx177v3 --params-file /tmp/launch_params_5ffcvn_3'].
```

### Steps to reproduce the bug

[office.tar.gz](https://github.com/open-rmf/rmf_ros2/files/10433286/office.tar.gz)

Extract and start the custom scenario

```bash
ros2 launch office/office.launch.xml
```

then send a patrol request

```bash
ros2 run rmf_demos_tasks dispatch_patrol -p patrol_A1_L1 patrol_A1_L2
```

### Expected behavior
Successfully plan a path, or fail gracefully if it cannot find one.

### Actual behavior
Crashes with no error messages.

Yadunund, 2023-01-17:
Upon investigating the building.yaml file, it seems
* Level fiducials were missing which are needed to derive the scale for the second level. Without this, waypoints on higher levels might have retained their pixel-coordinate values.
* several waypoint properties had their keys incorrectly modified (_L1 or _L2 were appended)
* Level `2F` had same elevation as `1F`

I've fixed these discrepancies and you can find the updated building yaml below.
[office_fixed.zip](https://github.com/open-rmf/rmf_ros2/files/10433869/office_fixed.zip)

However it will be good to fix a bug that surfaced in `GoToPlace` where we're not checking if battery_soc > 0 before updating the finish state.
https://github.com/open-rmf/rmf_task/blob/7bc64b63ea163e5f90c70f76b372102f27440678/rmf_task_sequence/src/rmf_task_sequence/events/GoToPlace.cpp#L142-L145

mxgrey, 2023-01-18:
Let's leave it open until the task planning issue is fixed. There are no scenarios where it's acceptable for the task planner (and more broadly, the fleet adapter) to crash.