open-rmf/rmf_ros2 #260: Issue with rmf_demos_tasks dispatch_patrol

HuzaifaOsal, 2023-01-18:
When i specify the name of robot we want to move in rmd_demo_tasks dispatch_patrol with -R flag the robot goes there without any issue 

command I am using :

_ros2 run rmf_demos_tasks dispatch_patrol -F tinyRobot -R tinyRobot1 -p point_18 -n 3 --use_sim_time_

but if we want openrmf to decide and dispatch task on it own. It does not work. Below i have pasted the output from the fleet_adapter

command I run:

_ros2 run rmf_demos_tasks dispatch_patrol -F tinyRobot -p point_18 -n 3 --use_sim_time_

_[rmf_traffic_schedule-1] [INFO] [1674032109.251925270] [rmf_traffic_schedule_primary]: [ScheduleNode::update_mirrors] Sending remedial update starting from the beginning going to 22 for query 1
[rmf_task_dispatcher-13] [INFO] [1674032132.873390487] [rmf_dispatcher_node]: Add Task [patrol.dispatch-0] to a bidding queue
[rmf_task_dispatcher-13] [INFO] [1674032132.927290661] [rmf_dispatcher_node]:  - Start new bidding task: patrol.dispatch-0
[full_control-15] [INFO] [1674032132.928143003] [abisysa_fleet_adapter]: [Bidder] Received Bidding notice for task_id [patrol.dispatch-0]
[full_control-15] [INFO] [1674032132.928456766] [abisysa_fleet_adapter]: Planning for [1] robot(s) and [1] request(s)
[full_control-15] [INFO] [1674032132.931429817] [abisysa_fleet_adapter]: Submitted BidProposal to accommodate task [patrol.dispatch-0] by robot [forklift_02] with new cost [51.765007]
[schedule_visualizer_node-4] [INFO] [1674032149.243155174] [schedule_data_node]: Requesting new schedule update because update timed out
[schedule_visualizer_node-4] [INFO] [1674032149.243250119] [schedule_data_node]: [rmf_traffic_ros2::MirrorManager::request_update] Requesting changes for query ID [1] since version [23]
[full_control-15] [INFO] [1674032149.243820402] [abisysa_fleet_adapter]: Requesting new schedule update because update timed out_

There is only one robot in the fleet tinyRobot1.
Please guide whats the problem here

Yadunund, 2023-01-18:
A similar issue has been reported in the past where the fleet adapter does not receive the `dispatch request` message published by the dispatcher. See https://github.com/open-rmf/rmf/discussions/171#discussioncomment-3231165

Could you share more information on your setup including
* RMW used
* Are you using Open-RMF binaries or a build from source? For the former, which version and for the latter you building the latest `main`? Let us know your OS version too and whether you're running this from a container such a Docker.

HuzaifaOsal, 2023-01-18:
OS: Ubuntu 22.04 
distro: ROS2 humble
yeah I build it from the latest main branch  
Our setup is we are using free_fleet_ros1_client  to communicate with our robots 
and free_fleet_server on ros2

HuzaifaOsal, 2023-01-18:
Thanks for pointing in right direction seems to work now.
It was clock sync issue between client and server 
Thank you

closed 2023-01-18 by HuzaifaOsal