open-rmf/rmf_ros2 #264: cancel_task not working

HuzaifaOsal, 2023-02-10:
Ubuntu 22.04
ROS2 Humble build from sources 

 trying to cancel task task from terminal and from rmf_web in both cases  when I execute in terminal
 _ros2 run rmf_demos_tasks cancel_task -id patrol.dispatch-1_

what I see message from fleet_adapter is 
_[fleet_adapter-16] [INFO] [1676024885.585866157] [tinyRobot_command_handle]: Received new path for tinyRobot1_
 
I see same ouput in case of canceling task from rmf_web 

 tried to call the cancel_task service but didint work 

whats the solution for this 

https://user-images.githubusercontent.com/117179645/218070644-4591d2c0-206b-4b4a-aad6-58d1429d7336.mp4

HuzaifaOsal, 2023-02-10:
is there a way to cancel the active task is the robot is currently execute

closed 2023-02-28 by HuzaifaOsal

Cavalletta98, 2024-07-12:
Hi @HuzaifaOsal , how did you solved it?

mxgrey, 2024-07-12:
You can use [`RobotUpdateHandle::current_task_id`](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/RobotUpdateHandle.hpp#L131) to get the current task ID.

Then you can pass that into [`RobotUpdateHandle::cancel_task`](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/RobotUpdateHandle.hpp#L319) to cancel the task.

There are Python bindings in `rmf_fleet_adapter_python` for both functions. 

You can also look into how task cancellation is done in https://github.com/open-rmf/rmf-web if you want to cancel using web hooks.