open-rmf/rmf_ros2 #271: Where is dock command trigger - fleet_adapter

mustafakurban, 2023-05-03:
Hello everyone, I hope I can explain my problem properly.

I want to add a new state to fleet_adapter, like dock. Actually the opposite of docking. I am using full control and I need to find where the dock command is triggered. This is the function I'm talking about (rmf_fleet_adapter/src/full_control/main.cpp)(line 320)
void dock(
    const std::string& dock_name,
    RequestCompleted docking_finished_callback) final
  {
    auto lock = _lock();
    _clear_last_command();

    _dock_finished_callback = std::move(docking_finished_callback);
    _current_dock_request.parameters.front().value = dock_name;
    _current_dock_request.task_id = std::to_string(++_current_task_id);

    _dock_requested_time = std::chrono::steady_clock::now();
    _mode_request_pub->publish(_current_dock_request);

    // TODO(MXG): We should come up with a better way to identify the docking
    // lanes.
    _dock_target_wp = rmf_utils::nullopt;
    DockFinder finder(dock_name);
    for (std::size_t i = 0; i < _travel_info.graph->num_lanes(); ++i)
    {
      const auto& lane = _travel_info.graph->get_lane(i);
      const auto entry_event = lane.entry().event();
      if (entry_event)
      {
        entry_event->execute(finder);
        if (finder.is_dock)
        {
          _dock_target_wp = lane.entry().waypoint_index();
          break;
        }
      }
    }

    assert(_dock_target_wp);

    const auto& wp = _travel_info.graph->get_waypoint(*_dock_target_wp);
    const std::string wp_name = wp.name() ?
      *wp.name() : std::to_string(wp.index());

    RCLCPP_INFO(
      _node->get_logger(),
      "Requesting robot [%s] of [%s] to dock into waypoint [%s]",
      _current_path_request.robot_name.c_str(),
      _current_path_request.fleet_name.c_str(),
      wp_name.c_str());
  }

Yadunund, 2023-05-17:
> I want to add a new state to fleet_adapter, like dock. Actually the opposite of docking.

It might be good if you can expand further on what you're trying to do. Maybe there's already a way to achieve the desired behavior without having to modify any source code.

Anyways, docking is a lane event that is triggered when a robot begins moving down a lane where the exit vertex of the lane is configured with a dock property.

When the task the fleet adapter beings executing has a `GoToPlace` phase in the task description, it will first create an [ExecutionPlan](https://github.com/open-rmf/rmf_ros2/blob/c8070a6ac6400545d683cd453fae27864e9b0a44/rmf_fleet_adapter/src/rmf_fleet_adapter/events/GoToPlace.cpp#L430-L432). This `ExecutionPlan` then determines the events that need to occurs along the path of the robot including any dock events. It then creates a phases for these events (eg for [dock}(https://github.com/open-rmf/rmf_ros2/blob/c8070a6ac6400545d683cd453fae27864e9b0a44/rmf_fleet_adapter/src/rmf_fleet_adapter/events/ExecutePlan.cpp#L105-L113) and [bundles](https://github.com/open-rmf/rmf_ros2/blob/c8070a6ac6400545d683cd453fae27864e9b0a44/rmf_fleet_adapter/src/rmf_fleet_adapter/events/ExecutePlan.cpp#L621-L623) all these phases in the right sequence in which they should execute. When it's time to execute a dock phase in the sequence, a [runtime phase is generated](https://github.com/open-rmf/rmf_ros2/blob/c8070a6ac6400545d683cd453fae27864e9b0a44/rmf_fleet_adapter/src/rmf_fleet_adapter/phases/DockRobot.cpp#L86-L89). Finally when this runtime phase is [triggered](https://github.com/open-rmf/rmf_ros2/blob/c8070a6ac6400545d683cd453fae27864e9b0a44/rmf_fleet_adapter/src/rmf_fleet_adapter/phases/DockRobot.hpp#L107) it calls the `RobotCommandHandle::dock()` function.

Yadunund, 2023-05-17:
Also please open such questions on our discussions board in the future: https://github.com/open-rmf/rmf/discussions

closed 2024-04-17 by mustafakurban