open-rmf/rmf_ros2 #279: Possible bugs caused by an empty plan generated by robots when acquiring new tasks

liugehaizaixue, 2023-06-02:
I have found that if a robot obtains a new task and needs to interrupt the current task to execute a new one, it may generate a bug

For example, when the target point of a new task received by a robot happens to be its current waypoint, an empty plan will be generated, and the prompt 'The planner indicators that the robot is already at its goal.'. At this point, the robot will not receive a new plan, then he will continue to execute the path of the previous task. This is a potential issue. Can we send an empty plan or cancel the robot's current plan when interrupting the current task

liugehaizaixue, 2023-06-03:
Should the current plan be discarded while canceling the task ? Is there a relevant interface or method ？

mxgrey, 2023-06-03:
By "current plan" do you mean the last path command that was given before you used the `interrupt` function? If so, that should always be discarded once you interrupt. Interrupting and then resuming a task while the robot is trying to reach a goal will always lead to a new plan being generated.

I'll double-check the code to see if it's correctly handling a situation where a task resumes when the robot is already at a location that it's supposed to go to.

liugehaizaixue, 2023-06-03:
> By "current plan" do you mean the last path command that was given before you used the `interrupt` function? If so, that should always be discarded once you interrupt. Interrupting and then resuming [...]

Actually, it's not 'interrupt', but I canceled the current task and issued a new task to take it to the current point. At this point, the generated 'plan' is empty, and the gotoplace will directly transition to a completed state without triggering ExecutePlan and MoveRobot. So fleet_adapter will not receive a new follow_ new_ path instruction, which will complete the remaing_waypoints issued by the previous task.

liugehaizaixue, 2023-06-03:
To add, if the task is manually cancelled and a new task is added, the above situation should not occur, as this automatically generates a  `responsive wait` during the interval to keep the robot at the current point.

However, if there is a task that can seize the current task when issued, that is, cancel the current task. At this point, the `taskmanager` will directly proceed to the next task `begin_next_task`, which will skip the interval of generating `responsive wait` and trigger the aforementioned bug.

mxgrey, 2023-06-06:
Thanks for elaborating, I misunderstood the problem description after seeing "interrupt" instead of "cancel".

> then he will continue to execute the path of the previous task

If a task is cancelled or killed while the robot is going to a place then the [`stop`](https://github.com/open-rmf/rmf_demos/blob/b71039d7d666ae999edcdd01ca271d3a87fd99fa/rmf_demos_fleet_adapter/rmf_demos_fleet_adapter/RobotCommandHandle.py#L217) command should be triggered on the robot's command handle. You should implement your command handle's `stop` function to clear out any information about the previous path command and have the robot stop wherever it currently is.

Let me know if you're not seeing the `stop` command get triggered when a task is cancelled while the robot is being commanded to go somewhere.

liugehaizaixue, 2023-06-07:
> If a task is cancelled or killed while the robot is going to a place then the [...]

Sorry, I just noticed that in the new version of the code, the 'interrupt' command has been added to the 'stop' command. This seems to have successfully solved the problem. Thank you.

closed 2023-06-07 by mxgrey