open-rmf/rmf_ros2 #287: [Bug]: Direct robot_task in DirectQueue is ordered wrongly

Magmanat, 2023-07-14:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

dc740df43cf34ac46730369eb528f38172798cb4

### ROS distribution

Humble

### ROS installation type

Source build

### Description of the bug

when sending multiple direct `robot_task` using rmf_web `/tasks/robot_task` api call, and with the `unix_millis_earliest_start_time` set as the time at which the task was sent. Queued missions are dispatched in descending order via [TaskManager::_begin_next_task()](https://github.com/open-rmf/rmf_ros2/blob/dc740df43cf34ac46730369eb528f38172798cb4/rmf_fleet_adapter/src/rmf_fleet_adapter/TaskManager.cpp#L1237) Due to [DirectQueue<set> being arranged in descending order](https://github.com/open-rmf/rmf_ros2/blob/dc740df43cf34ac46730369eb528f38172798cb4/rmf_fleet_adapter/src/rmf_fleet_adapter/TaskManager.hpp#L82). 

### Steps to reproduce the bug

post robot_task with different unix times and record the rmf_id that is assigned to them
let the tasks execute in succession and realise that it chooses the largest earliest start time first

### Expected behavior

The task that was actually sent first should be the one that is dispatched to the robot based on earliest unix time

### Actual behavior

The task that was sent last was the one the one that is dispatched to the robot based on earliest unix time

Yadunund, 2023-07-17:
Thanks for reporting the problem. Could you help confirm if this [PR](https://github.com/open-rmf/rmf_ros2/pull/288) resolves the issue?

Magmanat, 2023-07-26:
Seems like this makes the direct_queue work as intended, executing the task with the earliest `unix_millis_earliest_start_time`

Yadunund, 2023-07-26:
Thanks for finding the bug and confirming the fix works!