open-rmf/rmf_ros2 #314: [Bug]: Robot doesn't create lift session with open door

Vortex-TH, 2023-12-22:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04

### Open-RMF installation type

Binaries

### Open-RMF version or commit hash

0.0.1-1jammy.20231206.081430

### ROS distribution

Iron

### ROS installation type

Binaries

### Description of the bug

In our facility we have an agv lift that doesn't have a door on the first floor level, so in the Lift adapter API (using the lift adapter template), I set the door to always be open when the lift is currently at that floor, ready for the robot to move in or out.

This causes a problem, as the robot apparently expects a closed door and a lift session only is created when the door is initially closed. I could create a 'virtual door' that only 'opens' when there is a request, but 1. that doesn't represent the real state 2. requires the integration of some arbitrary triggers for door control depending on the robot

I'm not sure if this is already addressed in #310. Unfortunately I don't have the opportunity to test that, maybe someone can have a look at it.

### Steps to reproduce the bug

1. Launch RMF
2. Launch Lift Adapter with Lift API that is configured to have a door open by default (when lift is on that floor level)
3. Sent the robot to a waypoint on another floor level

### Expected behavior

The robot should request a Lift session and move into the lift

### Actual behavior

No lift session is created (the robot waits in front of the lift and nothing happens), even though the door is open.

Vortex-TH, 2024-01-19:
Also, lift sessions don't get terminated when there is a door that stays open (or no door at all) after reaching a floor level.

mxgrey, 2024-01-20:
Sorry for not noticing this issue report sooner.

I'm a bit perplexed by the problem that you're describing. I'm looking through the source code and I don't see anywhere that requires the lift doors to ever be closed.

> the robot apparently expects a closed door and a lift session only is created when the door is initially closed

Lift sessions are managed by the [lift supervisor](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/lift_supervisor/Node.cpp) which has no expectation of the door being open or closed; it just forwards whatever is asked for by the fleet adapters.

The fleet adapter only ever requests the [door to be opened](https://github.com/open-rmf/rmf_ros2/blob/efccf11088055e283a3a29071ee7a0364f405af6/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/RobotContext.cpp#L1388). The assumption is that the lift door will automatically close when it needs to, so there's no need for us to explicitly command a closed door.

The workflow for lift usage is only blocked on [waiting for the door to be open](https://github.com/open-rmf/rmf_ros2/blob/efccf11088055e283a3a29071ee7a0364f405af6/rmf_fleet_adapter/src/rmf_fleet_adapter/phases/RequestLift.cpp#L280). We don't have any condition that waits for the door to be closed.

Waiting for a lift session to end [doesn't check the door state at all](https://github.com/open-rmf/rmf_ros2/blob/efccf11088055e283a3a29071ee7a0364f405af6/rmf_fleet_adapter/src/rmf_fleet_adapter/phases/EndLiftSession.cpp#L104-L111).

I think there must be some other problem blocking the workflow for you. Could you recreate the issue and then run

```
$ ros2 topic echo lift_states
$ ros2 topic echo lift_requests
$ ros2 topic echo adapter_lift_requests
```

and let me know what messages come out of each?

Vortex-TH, 2024-01-20:
Thanks for the reply. I hope I can narrow down the issue next week if I have some free time. I'll have a look if I added something that may cause this. 

I remember there was some confusion about what to return for the value of [`destination_floor`](https://github.com/open-rmf/lift_adapter_template/blob/77ae1817db6cc2c165afeafb71a809c25effd579/lift_adapter_template/lift_adapter_template/LiftAPI.py#L81) when the lift is idle at the floor, but I'm not sure if that was causing issues. I'll check.

I'm also not sure if [the other issue](https://github.com/open-rmf/rmf_ros2/issues/315) I reported (and is reproducible in the sim) has an effect or is intended behaviour. Maybe you can have a look at that too.

Vortex-TH, 2024-01-22:
I noticed that there seems to be an empty session that blocks the lift request by the robot:
```
[lift_supervisor-12] [INFO] [1705944941.145228398] [rmf_lift_supervisor]: [] Published lift request to [L1] from lift supervisor
[fleet_adapter_neo-14] [INFO] [1705944941.145494490] [neo_fleet_adapter]: [neo/neo_2] is waiting to begin a session with lift [MyLift] but the lift is currently held by []
[lift_supervisor-12] [INFO] [1705944941.145883701] [rmf_lift_supervisor]: [neo/neo_2] Received adapter lift request to [L1] with request type [1]
[lift_supervisor-12] [INFO] [1705944942.145425284] [rmf_lift_supervisor]: [] Published lift request to [L1] from lift supervisor
[fleet_adapter_neo-14] [INFO] [1705944942.145653706] [neo_fleet_adapter]: [neo/neo_2] is waiting to begin a session with lift [MyLift] but the lift is currently held by []
[lift_supervisor-12] [INFO] [1705944942.145969155] [rmf_lift_supervisor]: [neo/neo_2] Received adapter lift request to [L1] with request type [1]
[lift_supervisor-12] [INFO] [1705944943.146061729] [rmf_lift_supervisor]: [] Published lift request to [L1] from lift supervisor
[fleet_adapter_neo-14] [INFO] [1705944943.146403358] [neo_fleet_adapter]: [neo/neo_2] is waiting to begin a session with lift [MyLift] but the lift is currently held by []
[lift_supervisor-12] [INFO] [1705944943.146671500] [rmf_lift_supervisor]: [neo/neo_2] Received adapter lift request to [L1] with request type [1]
[lift_supervisor-12] [INFO] [1705944944.145847326] [rmf_lift_supervisor]: [] Published lift request to [L1] from lift supervisor
[fleet_adapter_neo-14] [INFO] [1705944944.146274891] [neo_fleet_adapter]: [neo/neo_2] is waiting to begin a session with lift [MyLift] but the lift is currently held by []
[lift_supervisor-12] [INFO] [1705944944.146552614] [rmf_lift_supervisor]: [neo/neo_2] Received adapter lift request to [L1] with request type [1]
[lift_supervisor-12] [INFO] [1705944945.148897978] [rmf_lift_supervisor]: [] Published lift request to [L1] from lift supervisor
[fleet_adapter_neo-14] [INFO] [1705944945.149180348] [neo_fleet_adapter]: [neo/neo_2] is waiting to begin a session with lift [MyLift] but the lift is currently held by []
[lift_supervisor-12] [INFO] [1705944945.149544510] [rmf_lift_supervisor]: [neo/neo_2] Received adapter lift request to [L1] with request type [1]
```

`/lift_requests` and `/adapter_lift_requests` look like this (published at 1Hz):
```
---
lift_name: MyLift
request_time:
  sec: 1705944943
  nanosec: 146480901
session_id: neo/neo_2
request_type: 1
destination_floor: L1
door_state: 2
---
```
or this:

```
---
lift_name: MyLift
request_time:
  sec: 1705945109
  nanosec: 452245532
session_id: ''
request_type: 0
destination_floor: L1
door_state: 0
---
```

`/lift_states`:
```
---
lift_time:
  sec: 1705945570
  nanosec: 316430784
lift_name: MyLift
available_floors:
- L1
- L2
current_floor: L1
destination_floor: L1
door_state: 2
motion_state: 0
available_modes:
- 1
- 2
current_mode: 2
session_id: ''
---

```

Vortex-TH, 2024-08-16:
So, I did some more experiments (on real hardware) with a clean approach and captured some logs.

As for the setup, my lift now looks like this: 

[ L3 ]
[ L2 ]
[ L1 ] <-- No door (= always open)

But for the tests, I only used L1 and L2.

For each test, I captured the following topics:

/lift_states
/lift_requests
/adapter_lift_requests

and also the output of my lift adapter.

<details>
<summary>Test 1: Idle lift in L1</summary>

---
[**/lift_states**](https://gist.github.com/Vortex-TH/9b8918069b1b4f72193ae5e07ef3e9c3)

[**/lift_requests**](https://gist.github.com/Vortex-TH/cbbf83326105692b182bcb8ff752534e)

**/adapter_lift_requests**
empty

**Liftadapter output**
Forgot to save output, but probably looks similiar to Test 2 :)

---
</details>

<details>
<summary>Test 2: Idle lift in L2</summary>

---
[**/lift_states**](https://gist.github.com/Vortex-TH/3001ab285f4672e3488c1009b27beab2)

[**/lift_requests**](https://gist.github.com/Vortex-TH/ef5eab8685185926af1c5dbd91af0de6)

**/adapter_lift_requests**
empty

**Liftadapter output**
```
[INFO] [1723802857.968736468] [lift_adapter_template]: Checking connectivity.
Elevator client connected.
[INFO] [1723802858.010080737] [lift_adapter_template]: Initializing with status.
Initial state: rmf_lift_msgs.msg.LiftState(lift_time=builtin_interfaces.msg.Time(sec=1723802858, nanosec=10292734), lift_name='MyLift', available_floors=['L1', 'L2', 'L3'], current_floor='L2', destination_floor='L2', door_state=<DoorState.OPEN: 2>, motion_state=<MotionState.STOPPED: 0>, available_modes=[1, 2], current_mode=2, session_id='')
[INFO] [1723802858.227493242] [lift_adapter_template]: Running LiftAdapterTemplate
[INFO] [1723802859.453635121] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802860.455794868] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802861.446050072] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802862.450716855] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802863.463305613] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802864.456636370] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802865.451086139] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802866.456297985] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802867.451232224] [lift_adapter_template]: Requested lift to L2.
[INFO] [1723802868.461929042] [lift_adapter_template]: Requested lift to L2.
[KeyboardInterrupt]
```

---
</details>

<details>
<summary>Test 3: Idle lift in L2, then request to L1 by Rviz panel</summary>

---
[**/lift_states**](https://gist.github.com/Vortex-TH/7b9ef65f8a8229624906cf0be55e21b3)

[**/lift_requests**](https://gist.github.com/Vortex-TH/694cd4d1f61bead5d6fd00d1683e4dc7)

**/adapter_lift_requests**
```
lift_name: MyLift
request_time:
  sec: 1723803591
  nanosec: 446226334
session_id: lift_panel_session
request_type: 1
destination_floor: L1
door_state: 2
---
[KeyboardInterrupt]
```

[**Liftadapter output**](https://gist.github.com/Vortex-TH/dfd1286865e8021e5a20a4691f212b2c)

**Observation**
The lift moves from L2 to L1 as requested from the lift panel in Rviz.

---
</details>

<details>
<summary>Test 4: Idle lift in L2, then request by AGV from L1</summary>

---
[**/lift_states**](https://gist.github.com/Vortex-TH/39055745a9eb1fdd6a672277d0615fcc)

[**/lift_requests**](https://gist.github.com/Vortex-TH/0cfbce24f5d89dc8c3d035a967c40161)

[**/adapter_lift_requests**](https://gist.github.com/Vortex-TH/e2349dee4dca907b09b28dbd2e73120b)

[**Liftadapter output**](https://gist.github.com/Vortex-TH/03ff0b20b2e0f6238684930294350c05)

**Observation**
The lift moves from L2 to L1, the AGV moves in to the lift, which then moves to L2.
The AGV then doesn't leave the lift, but that may be another issue caused by a faulty localization or something which I will address later.

---
</details>

<details>
<summary>Test 5: Idle lift in L1, then request by AGV from L1</summary>

---
[**/lift_states**](https://gist.github.com/Vortex-TH/18ee5dad8a09c11c64bd0726ae9ef559)

[**/lift_requests**](https://gist.github.com/Vortex-TH/7e8b4bee4a3b68137b193d5edf4d48ef)

**/adapter_lift_requests**
```
lift_name: MyLift
request_time:
  sec: 1723810271
  nanosec: 677284008
session_id: neo/neo_1
request_type: 1
destination_floor: L1
door_state: 2
---
[KeyboardInterrupt]
```

[**Liftadapter output**](https://gist.github.com/Vortex-TH/9d6c386f875224dd9e2f2156e80956fe)

**Observation**
The AGV moves in front of the lift, but then nothing happens, although the lift is ready for it to move in.

---
</details>

**Results**

I think Test 1 to 4 worked as they should, although I may have noticed a problem in Test 1 which could have caused Test 5 to fail:

When the lift is in L1, RMF repeatedly requests it to L1 with door_state 0 (closed), which is never going to be achieved since the door is always open. Could this have the effect that there is a session that doesn't get released and thus the AGV doesn't move into the lift?

xiyuoh, 2024-08-19:
Hello @Vortex-TH ! Thank you for the logs, they're very helpful to understand what is going on with these lift sessions. I just wanted to clarify about Tests 4 and 5:

Test 4:
Is the objective of Test 4 **(a)** to see if the AGV can command the lift to enter, then end the lift session, or **(b)** to see if the AGV can travel from L1 to L2?
I notice that the lift is being requested from L2 to L1 by the AGV so that the AGV could enter. May I check if there was a request by the user for the lift to go to L2 after the robot has entered the lift, or did the lift request to L2 come out of nowhere? Mainly curious about where [these LiftRequests](https://gist.github.com/Vortex-TH/0cfbce24f5d89dc8c3d035a967c40161#file-lift_requests-L820-L836) came from.

Test 5:
What are the logs on RMF fleet adapter when the lift is ready and doors opened on L1 for the robot to move in?

Vortex-TH, 2024-08-19:
> Test 4: Is the objective of Test 4 **(a)** to see if the AGV can command the lift to enter, then end the lift session, or **(b)** to see if the AGV can travel from L1 to L2? 

I want to see whether the AGV goes from L1 to L2, which it did.

> I notice that the lift is being requested from L2 to L1 by the AGV so that the AGV could enter. May I check if there was a request by the user for the lift to go to L2 after the robot has entered [...]

That would be the request by the AGV to move to L2 from L1 after it has moved in. No manual intervention was done in this test.

> Test 5: What are the logs on RMF fleet adapter when the lift is ready and doors opened on L1 for the robot to move in?

I didn't save the output, so I will test it again and see. I think it looks like I posted in my previous comment, but I'll check.

xiyuoh, 2024-08-19:
Thanks for clarifying! I just spent some time tracing the logs, and the requests provided seem very incoherent. I've mapped them out in the chart below.

[image]

Commands with white text will be ignored as RMF takes into account the request time and ensures that it was not a past request. 

To explain a little about how these lift requests work, the various fleet adapters will publish their lift requests via `/adapter_lift_requests` to the lift supervisor. The lift supervisor will then process these requests one robot at a time, i.e. only entertain requests from one session id (which usually represents the robot in the `fleet_name/robot_name` convention) on a first-come-first-serve basis. These filtered requests for the single robot will then be published to the lift adapter on the `/lift_requests` topic.

You can see from the image that there are some rogue requests being sent on `/lift_requests` topic that did not come from `/adapter_lift_requests`. Even though Test 4 seems to behave as it should, I think resolving why this happened may help us with finding out why Test 5 failed and any hidden bugs in our lift request implementation.

To further isolate the issue, I suggest running the same tests after doing the following:
- `sudo apt update && sudo apt upgrade` to use the latest binaries
- Check that each of these tests are done on a clean setup, i.e. no stray nodes running in the background from a previous test.
- Ensure that we are not receiving inaccurate states and requests from previous runs with `ros2 topic info /lift_requests -v` and `ros2 topic info /lift_states -v`.

Let me know if you're still facing the same problem after these steps. In the meantime I will try to reproduce this in simulation.

xiyuoh, 2024-08-20:
I thought more about Test 5 and I noticed that after a request is published on `/lift_requests` with the [session ID](https://gist.github.com/Vortex-TH/7e8b4bee4a3b68137b193d5edf4d48ef#file-lift_requests-L185) `neo/neo_1`, the [session ID](https://gist.github.com/Vortex-TH/18ee5dad8a09c11c64bd0726ae9ef559#file-lift_states-L359) published on `/lift_states` never got updated. That is probably why the robot didn't move in; the session ID did not match the robot name so RMF thinks that the lift was busy with another session. Could you check that your lift adapter is publishing the lift states and populating the session id?

xiyuoh, 2024-08-21:
Unfortunately I haven't been able to reproduce this in simulation with 22.04 + RMF Iron binaries. Some observations about your latest tests:
- It is odd that the lift supervisor is the one [ending the lift session](https://gist.github.com/Vortex-TH/5d9a1c25cf295ae5f4a1d91e6b05dbf8#file-rmf-nodes-output-L50-L55) after having requested for it.
- I previously mentioned that RMF will ignore [messages without timestamps](https://gist.github.com/Vortex-TH/43d305e27a99e6f8d1a5803e4b6d20fb#file-adapter_lift_requests-L12-L13), but it is strange that these are being published in the first place.
- There are messages published on `/lift_requests` with the same timestamp [16 times](https://gist.github.com/Vortex-TH/35d0b7de27f97a458aae4dfb8f645813#file-lift_requests-L739-L881)

I'm really interested to understand where these stray `END_SESSION` requests are coming from so that we can fix existing bugs. I'd need your help gathering a little more info:

1. You mentioned that the packages are up to date, can I check what is the output when you do a `apt list --installed | grep iron-rmf`? For reference when I do that, the packages are dated 20240713.
2. Can I also ask for an output of `ros2 topic info /adapter_lift_requests -v`?
3. To ensure that we're capturing all the info we can get, do you think you can do a screen recording like those in this [ticket](https://github.com/open-rmf/rmf_ros2/issues/382#issuecomment-2298041977)? It'd be helpful to see:
   - `ros2 topic echo /lift_requests`
   - `ros2 topic echo /adapter_lift_requests`
   - `ros2 topic echo /lift_states`
   - The terminal you use to launch the fleet adapter
4. I should have asked this earlier, but may I know what type of setup are you testing with? I saw that you tried this on real hardware - does that mean real robots + real lift, or real robots + simulated lift? Are you using docker for RMF?

Vortex-TH, 2024-08-21:
I just realized that the sourced rmf was some compiled version from an earlier test instead of the binary.

Sorry about that. I will repeat the test and make sure it's the correct version.

Vortex-TH, 2024-08-21:
Ok, first the requested data:

<details>
<summary>apt list --installed | grep iron-rmf</summary>

```
ros-iron-rmf-api-msgs/jammy,now 0.1.2-1jammy.20240713.181510 amd64 [installed,automatic]
ros-iron-rmf-battery/jammy,now 0.2.1-1jammy.20240711.223637 amd64 [installed,automatic]
ros-iron-rmf-building-map-msgs/jammy,now 1.3.1-1jammy.20240711.233958 amd64 [installed,automatic]
ros-iron-rmf-building-map-tools/jammy,now 1.7.2-1jammy.20240713.181700 amd64 [installed,automatic]
ros-iron-rmf-building-sim-common/jammy,now 2.1.2-1jammy.20240713.181906 amd64 [installed,automatic]
ros-iron-rmf-building-sim-gz-classic-plugins/jammy,now 2.1.2-1jammy.20240713.182226 amd64 [installed,automatic]
ros-iron-rmf-building-sim-gz-plugins/jammy,now 2.1.2-1jammy.20240713.182226 amd64 [installed,automatic]
ros-iron-rmf-charger-msgs/jammy,now 3.1.3-1jammy.20240713.181517 amd64 [installed,automatic]
ros-iron-rmf-dev/jammy,now 0.0.1-1jammy.20240731.202307 amd64 [installed]
ros-iron-rmf-dispenser-msgs/jammy,now 3.1.3-1jammy.20240713.181645 amd64 [installed,automatic]
ros-iron-rmf-door-msgs/jammy,now 3.1.3-1jammy.20240713.181513 amd64 [installed,automatic]
ros-iron-rmf-fleet-adapter-python/jammy,now 2.2.6-1jammy.20240713.191142 amd64 [installed,automatic]
ros-iron-rmf-fleet-adapter/jammy,now 2.2.6-1jammy.20240713.184205 amd64 [installed,automatic]
ros-iron-rmf-fleet-msgs/jammy,now 3.1.3-1jammy.20240713.181539 amd64 [installed,automatic]
ros-iron-rmf-ingestor-msgs/jammy,now 3.1.3-1jammy.20240713.182215 amd64 [installed,automatic]
ros-iron-rmf-lift-msgs/jammy,now 3.1.3-1jammy.20240713.181641 amd64 [installed,automatic]
ros-iron-rmf-obstacle-msgs/jammy,now 3.1.3-1jammy.20240713.181528 amd64 [installed,automatic]
ros-iron-rmf-robot-sim-common/jammy,now 2.1.2-1jammy.20240731.185003 amd64 [installed,automatic]
ros-iron-rmf-robot-sim-gz-classic-plugins/jammy,now 2.1.2-1jammy.20240731.192754 amd64 [installed,automatic]
ros-iron-rmf-robot-sim-gz-plugins/jammy,now 2.1.2-1jammy.20240731.192800 amd64 [installed,automatic]
ros-iron-rmf-scheduler-msgs/jammy,now 3.1.3-1jammy.20240713.181536 amd64 [installed,automatic]
ros-iron-rmf-site-map-msgs/jammy,now 3.1.3-1jammy.20240713.181526 amd64 [installed,automatic]
ros-iron-rmf-task-msgs/jammy,now 3.1.3-1jammy.20240713.182453 amd64 [installed,automatic]
ros-iron-rmf-task-ros2/jammy,now 2.2.6-1jammy.20240713.183707 amd64 [installed,automatic]
ros-iron-rmf-task-sequence/jammy,now 2.2.5-1jammy.20240713.181650 amd64 [installed,automatic]
ros-iron-rmf-task/jammy,now 2.2.5-1jammy.20240711.223755 amd64 [installed,automatic]
ros-iron-rmf-traffic-editor-assets/jammy,now 1.7.2-1jammy.20240711.215742 amd64 [installed,automatic]
ros-iron-rmf-traffic-editor-test-maps/jammy,now 1.7.2-1jammy.20240713.181844 amd64 [installed,automatic]
ros-iron-rmf-traffic-editor/jammy,now 1.7.2-1jammy.20240711.225642 amd64 [installed,automatic]
ros-iron-rmf-traffic-examples/jammy,now 3.1.2-1jammy.20240711.223701 amd64 [installed,automatic]
ros-iron-rmf-traffic-msgs/jammy,now 3.1.3-1jammy.20240713.181543 amd64 [installed,automatic]
ros-iron-rmf-traffic-ros2/jammy,now 2.2.6-1jammy.20240713.182409 amd64 [installed,automatic]
ros-iron-rmf-traffic/jammy,now 3.1.2-1jammy.20240711.221356 amd64 [installed,automatic]
ros-iron-rmf-utils/jammy,now 1.5.1-1jammy.20240711.221314 amd64 [installed,automatic]
ros-iron-rmf-visualization-building-systems/jammy,now 2.1.2-1jammy.20240713.181907 amd64 [installed,automatic]
ros-iron-rmf-visualization-fleet-states/jammy,now 2.1.2-1jammy.20240713.182014 amd64 [installed,automatic]
ros-iron-rmf-visualization-floorplans/jammy,now 2.1.2-1jammy.20240712.003540 amd64 [installed,automatic]
ros-iron-rmf-visualization-msgs/jammy,now 1.3.1-1jammy.20240711.232701 amd64 [installed,automatic]
ros-iron-rmf-visualization-navgraphs/jammy,now 2.1.2-1jammy.20240713.183707 amd64 [installed,automatic]
ros-iron-rmf-visualization-obstacles/jammy,now 2.1.2-1jammy.20240713.181839 amd64 [installed,automatic]
ros-iron-rmf-visualization-rviz2-plugins/jammy,now 2.1.2-1jammy.20240731.201438 amd64 [installed,automatic]
ros-iron-rmf-visualization-schedule/jammy,now 2.1.2-1jammy.20240713.183706 amd64 [installed,automatic]
ros-iron-rmf-visualization/jammy,now 2.1.2-1jammy.20240731.202219 amd64 [installed,automatic]
ros-iron-rmf-websocket/jammy,now 2.2.6-1jammy.20240713.181547 amd64 [installed,automatic]
ros-iron-rmf-workcell-msgs/jammy,now 3.1.3-1jammy.20240713.181549 amd64 [installed,automatic]
```
</details>

<details>
<summary>ros2 topic info /adapter_lift_requests -v</summary>

```
Type: rmf_lift_msgs/msg/LiftRequest

Publisher count: 2

Node name: lift_panel_session_node
Node namespace: /
Topic type: rmf_lift_msgs/msg/LiftRequest
Topic type hash: RIHS01_e8f1e0aa4d172703d0d7840c1876448c6c5d75ab58955faf7de7dddc17c096ac
Endpoint type: PUBLISHER
GID: 01.10.01.f2.cc.dc.a9.8e.36.ed.9d.c9.00.00.62.03
QoS profile:
  Reliability: RELIABLE
[... 37 lines cut ...]
  Liveliness lease duration: Infinite

```
</details>

Robots and lift are real hardware and I'm not using RMF in docker. I now also made sure I'm using the binary installation of RMF, but Test 4.4 and 5.2 still failed with the same problem (robot doesn't move into the lift).

<details>
<summary>Test 4.4: Idle lift in L2, then request by AGV from L1</summary>

[test4-4.webm](https://github.com/user-attachments/assets/3baa66b7-77cf-4ff2-9f28-6e3ecf0f5970)

</details>
<details>
<summary>Test 4.5: Idle lift in L2, then request by AGV from L1</summary>

[test4-5.webm](https://github.com/user-attachments/assets/f232e134-abf7-46d5-8036-7101a19b1db2)

</details>
<details>
<summary>Test 5.2: Idle lift in L2, then request by AGV from L1</summary>

[test5-2.webm](https://github.com/user-attachments/assets/0322a036-50af-490f-87a1-6f823807963e)

</details>

xiyuoh, 2024-08-22:
Thank you for providing the requested data! I see that the random `END_SESSION` requests are gone now, that a good sign!

For the remaining issue of why your robot does not enter the lift even when it reached L1 with doors open, I believe it all comes down to a mismatch in session ID. It would help to understand if you made any edits to the [lift adapter template](https://github.com/open-rmf/lift_adapter_template/tree/main) (assuming you are using this) for us to figure out how to correct this behavior.

**Test 4.4**
I can see why the robot does not enter the lift. Once the lift reached L1, the session ID changed from `neo/neo_1` to `""` (at the 1:26 mark of the video), even though there were no `END_SESSION` requests from the lift request topics. I suggest checking the lift adapter implementation and ensuring that the lift's session ID is not modified unless the lift adapter receives an `END_SESSION` request. Specifically you might want to check if [this logic](https://github.com/open-rmf/lift_adapter_template/blob/main/lift_adapter_template/lift_adapter_template/lift_adapter_template.py#L121-L126) is untouched in your lift adapter.

**Test 4.5**
Same problem as above, but it worked because there was _one_ single message published on `/lift_states` where the lift is at L1 and the session ID was still `neo/neo_1`, and therefore RMF allowed the robot to move into the lift. Immediately after that, the session ID becomes `""` again. For reference these happened at the 1:24 mark.

**Test 5.2**
This would be a repeat of the [previous issue](https://github.com/open-rmf/rmf_ros2/issues/314#issuecomment-2298091479) where the session ID was never updated on MyLift. We can see on `/lift_states` that the session ID stay `""` throughout the video.

Vortex-TH, 2024-08-22:
I am using [this version ](https://github.com/open-rmf/lift_adapter_template/tree/77ae1817db6cc2c165afeafb71a809c25effd579) of lift_adapter_template without any changes to lift_adapter_template.py.

Vortex-TH, 2024-08-22:
I added some quick print debugging around that adapter logic:

```
        print()
        print("lift_request:")
        print(self.lift_request)

        if self.lift_request is not None:
            print("Lift request is not None")
            print("and session_id is", self.lift_request.session_id)
            if self.lift_request.request_type == \
                    LiftRequest.REQUEST_END_SESSION:
                print("Lift request is REQUEST_END_SESSION")
                new_state.session_id = ''
            else:
                print("Lift request is not REQUEST_END_SESSION, setting new_state.session_id")
                new_state.session_id = self.lift_request.session_id
        print("Returning new_state:")
        print(new_state)
        print()
        return new_state

```

```
lift_request:
rmf_lift_msgs.msg.LiftRequest(lift_name='MyLift', request_time=builtin_interfaces.msg.Time(sec=1724330294, nanosec=333707105), session_id='neo/neo_1', request_type=1, destination_floor='L1', door_state=2)
Lift request is not None
and session_id is neo/neo_1
Lift request is not REQUEST_END_SESSION, setting new_state.session_id
Returning new_state:
rmf_lift_msgs.msg.LiftState(lift_time=builtin_interfaces.msg.Time(sec=1724330295, nanosec=636705808), lift_name='MyLift', available_floors=['L1', 'L2', 'L3'], current_floor='L1', destination_floor='L1', door_state=<DoorState.OPEN: 2>, motion_state=<MotionState.STOPPED: 0>, available_modes=[1, 2], current_mode=2, session_id='neo/neo_1')

lift_request:
None
Returning new_state:
rmf_lift_msgs.msg.LiftState(lift_time=builtin_interfaces.msg.Time(sec=1724330296, nanosec=136698010), lift_name='MyLift', available_floors=['L1', 'L2', 'L3'], current_floor='L1', destination_floor='L1', door_state=<DoorState.OPEN: 2>, motion_state=<MotionState.STOPPED: 0>, available_modes=[1, 2], current_mode=2, session_id='')

[INFO] [1724330296.347279582] [lift_adapter_template]: Requested lift to L1.

lift_request:
rmf_lift_msgs.msg.LiftRequest(lift_name='MyLift', request_time=builtin_interfaces.msg.Time(sec=1724330295, nanosec=332564131), session_id='neo/neo_1', request_type=1, destination_floor='L1', door_state=2)
Lift request is not None
and session_id is neo/neo_1
Lift request is not REQUEST_END_SESSION, setting new_state.session_id
Returning new_state:
rmf_lift_msgs.msg.LiftState(lift_time=builtin_interfaces.msg.Time(sec=1724330296, nanosec=636611981), lift_name='MyLift', available_floors=['L1', 'L2', 'L3'], current_floor='L1', destination_floor='L1', door_state=<DoorState.OPEN: 2>, motion_state=<MotionState.STOPPED: 0>, available_modes=[1, 2], current_mode=2, session_id='neo/neo_1')

```

It looks like it receives some callbacks with NoneType objects where it sets the session_id to ''

xiyuoh, 2024-08-23:
Ah I see what's going on, I suspect it is due to the logic inside the lift adapter template. I've made some changes to the template code [here](https://github.com/open-rmf/lift_adapter_template/commit/973adb29b473b037f033a9a003fd4b06d7cd4658) on the [xiyu/reset_request](https://github.com/open-rmf/lift_adapter_template/tree/xiyu/reset_request) branch, do you mind applying these changes and let me know if you're still experiencing the same issue?

Vortex-TH, 2024-08-23:
Sure, but I won't be able to test it until sometime next week.

Vortex-TH, 2024-08-28:
I tested your latest version and it looks like the robot is now always moving into the lift in all tests, but then the lift does not start moving.

[test5-3.webm](https://github.com/user-attachments/assets/88ad2133-47de-4376-97d7-e46af777a02c)

xiyuoh, 2024-08-28:
Glad to know that the robot moves into the lift now.

I see that on `/lift_states` the `destination_floor` is not being updated, similar to the previous issue where your `session_id` wasn't updated correctly. Could you check your lift implementation and ensure that the `destination_floor` is updating as it should (according to the valid lift requests with the correct session ID)?

xiyuoh, 2024-08-28:
@Vortex-TH I wanted to add that the [PR](https://github.com/open-rmf/lift_adapter_template/pull/4) was just updated today after a round of review, do ensure that your code is updated with the latest changes!

Vortex-TH, 2024-08-28:
I don't quite understand.. The destination floor should be updated after the lift received the command_lift(L2) callback, right? But the lift adapter never receives that function call (because it's busy with another request as seen in the video?).

> @Vortex-TH I wanted to add that the https://github.com/open-rmf/lift_adapter_template/pull/4 was just updated today after a round of review, do ensure that your code is updated with the latest changes!

Yes, I'm using the version from today

xiyuoh, 2024-08-28:
@Vortex-TH Thanks for pointing that out! I just updated the PR, I believe the lift adapter was rejecting all new requests as long as it is in the middle of a lift session. I've added a condition to only reject lift requests with different session IDs, will have to trouble you to try the latest code out.

Vortex-TH, 2024-08-28:
It seems to be working now! Thanks for your help :)

xiyuoh, 2024-08-28:
That's awesome, thanks for testing it! If there's nothing else I'll close this ticket, feel free to open a new one if you face a different issue.

closed 2024-08-28 by xiyuoh