open-rmf/rmf_ros2 #315: [Bug]: Lift functionality broken since latest update

Vortex-TH, 2024-01-03:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04

### Open-RMF installation type

Binaries

### Open-RMF version or commit hash

0.0.1-1jammy.20231226.151101

### ROS distribution

Iron

### ROS installation type

Binaries

### Description of the bug

[Since the latest update](https://github.com/open-rmf/rmf/releases/tag/release-iron-231229) I have problems with the lift adapter functionality. For some reason RMF continuously triggers lift requests, even though there is no AGV and the lift is already at that floor level, making it impossible to manually control the lift.

This doesn't seem right, does it?

[image]

I have this issue with my real lift and could also reproduce it with the hotel demo simulation, which worked before.

### Steps to reproduce the bug

Launch rmf with a map that has a lift, for example the hotel demo.

Vortex-TH, 2024-01-03:
Nevermind, it seems that the lift actually still works. However, I'm still wondering what all these lift requests are about.

mxgrey, 2024-01-20:
Thanks for pointing this out. There shouldn't be any functional issues. The simulated lifts should be working as normal, they're just being more noisy than they used to because the lift supervisor used to only publish lift requests on an as-needed basis but now the requests are being published periodically. The simulated lift wasn't originally programmed to receive periodic request messages so it created more noise than intended.

I've opened a PR that fixes the noise, at least for lifts that are not in use: https://github.com/open-rmf/rmf_simulation/pull/115

The new periodic request publishing is not an ideal or a final design choice. It was found to be necessary for a deployment where the lift adapter sometimes needed to be stopped and restarted, and when that happens it gets flooded with old outdated lift requests, e.g. requests that might have been sent yesterday. Those old requests would start sending the lift to the wrong floor needlessly. Various strategies were considered for filtering out old requests, but each strategy had some kind of critical flaw.

We're having an internal discussion about how to improve the situation and get rid of the periodic publishing in the next release, but for now it shouldn't be detrimental to behavior; it just creates a little bit more network traffic.

cardboardcode, 2024-08-15:
+1. Encountered a similar behaviour in my own setup when there is **continuous spam of lift requests** from RMF to Lift Adapter when starting RMF without an AGV. 

The lift requests I am getting are specifically all `LiftRequest.REQUEST_END_SESSION`

## Environment
**OS**: `Ubuntu 22.04`
**Open-RMF installation type**: `Binaries`
**Open-RMF version or commit hash**: [Open-RMF on Humble Hawksbill - Sync (2023-12-29)](https://github.com/open-rmf/rmf/releases/tag/release-humble-231229)
**ROS distribution**: `ROS 2 Humble`
**ROS installation type**: `Binaries`

The output from ROS 2 topic `/lift_requests` are as such:

```bash
---
lift_name: A
request_time:
  sec: 1723703818
  nanosec: 655260483
session_id: ''
request_type: 0
destination_floor: L2
door_state: 0
---
lift_name: B
request_time:
  sec: 1723703818
  nanosec: 655437758
session_id: ''
request_type: 0
destination_floor: L1
door_state: 0
---
...
```
Observed that the `destination floor` is always set to the `current_floor` of the respective lift.

This spam seems to be interfering with actual lift requests made to move the lift to user-selected floors as any sessions are requested to end immediately.

xiyuoh, 2024-08-16:
Hello @cardboardcode ! In order for me to reproduce your issue and locate any bugs, may I get a better understanding about what you're experiencing:
- Are you testing this on hardware or sim? If sim, can I check if you're using any Gazebo worlds? What is your Real-Time Factor?
- You mentioned that the spam is interfering with actual lift requests, what is happening with the lifts? Here's the flow of events I understand from your description, feel free to correct any inaccuracy:
   1. Lift is doing nothing, and on `/lift_requests` there is continuous stream of `END_SESSION` requests coming in (which is an intended RMF behavior)
   2. You submit a manual lift request for Lift A to L1.

   I'm unclear what happens after that. Does your request get rejected/ignored? What is the output when you do a `ros2 topic echo /lift_states` after Step ii? What are the logs printed?
- You provided some output from the topic `/lift_requests`, are these the output retrieved at Step i or after Step ii? If they are from Step i, may I see the output of this topic after Step ii?

cardboardcode, 2024-08-16:
@xiyuoh Thank you for the prompt response.

Will get back to you latest by **next Tuesday (20th August)** with the requested information. For now, will respond to those I can.
:question: 
> Are you testing this on hardware or sim? If sim, can I check if you're using any Gazebo worlds? What is your Real-Time Factor?

:arrow_right: Testing on simulation. Specifically, RMF with simulated KONE lift adapter. Both running within their respective docker containers.

:question: 
> You mentioned that the spam is interfering with actual lift requests, what is happening with the lifts? Here's the flow of events I understand from your description, feel free to correct any inaccuracy:

:arrow_right: Your assessment is correct. What happens after is that the manual lift request fails since it is immediately followed by an End Session lift request from the spam. This failure is reported in a websocket response after the API call is made for the manual request.

Will properly record a video showing what is outputted on `/lift_states` and `/lift_requests` as it is running. 

:question: 
> You provided some output from the topic /lift_requests, are these the output retrieved at Step i or after Step ii? If they are from Step i, may I see the output of this topic after Step ii?

:arrow_right: They are retrieved at **Step i**. They also show similar outputs with `request_type = 0` after **Step ii**.  Will include it in the video I will recording. 

Also do let me know **if you prefer I open a new issue** for this since it may start deviating from the original scope of the initial issue in this thread.

Vortex-TH, 2024-08-16:
I also did some more tests and uploaded a few logs to my [other issue report](https://github.com/open-rmf/rmf_ros2/issues/314#issuecomment-2293893766)). I don't know if it's related to the issues in this thread (and if they are caused by the lift request spam), but maybe the logs are helpful.

xiyuoh, 2024-08-19:
Thanks both for the info, it is definitely more helpful!

@cardboardcode Gotcha about the lift request failure. Looking forward to the `/lift_states` and `/lift_requests` outputs 😄 I missed out requesting for the logs from the lift adapter too, would appreciate if we could take a look at those! And yes, I think we can open a new ticket for further discussion.

@Vortex-TH thanks for highlighting your other issue report! I suspect that it might be a different issue from @cardboardcode as you managed to have the lift move to different levels via both rviz panel and AGV request, so I'll look into them separately first.

May I check for both of you, when you implement the lift adapter, how does the lift adapter determine whether a command fails (i.e. it returns False [here](https://github.com/open-rmf/lift_adapter_template/blob/main/lift_adapter_template/lift_adapter_template/LiftAPI.py#L105-L112))? Does it depend on a specific failure message from your lift, or does it return False if there's no response from the lift?

cardboardcode, 2024-08-20:
Hi @xiyuoh, regarding the issue of continuous `LiftRequest.REQUEST_END_SESSION` spam, **figured out the reason why (partially).**

## **Cause Of Issue** :mag: 
Was using an outdated version of `ghcr.io/open-rmf/rmf/rmf_demos:latest` as a docker image base for the `rmf_schedule_node` I was running dockerised. 

The image base was still based in ROS 2 Humble before it was updated to ROS 2 Jazzy a few weeks ago. 

:warning: Still unsure why this version of RMF would spam the way it did. 

## **Solution** :star:  
To resolve this, I build a new docker image with the base image being `ros:humble-ros-core` with the binaries of `ros-humble-rmf-dev` installed. 
With this new docker container running `rmf_schedule_node`, the spam stopped. 

## **Remarks** :speech_balloon: 
Will continue to create a new thread to capture this issue since I am still able to reproduce it. Would also like to know more about why it still behaved the way it did.

cardboardcode, 2024-08-20:
❓
> May I check for both of you, when you implement the lift adapter, how does the lift adapter determine whether a command fails (i.e. it returns False [...]

➡️ Regarding this enquiry, the Lift Adapter I am using in my setup unfortunately deviates from the `lift_adapter_template` and therefore does not have the `command_lift` function.

The Lift Adapter is `chart_sg/lift_kone_adapter` and can be accessed [here](https://github.com/KABAM-Robotics/lift_kone_adapter/tree/feature/update_dockerfile) for more details.

xiyuoh, 2024-08-20:
I can share a little about why there was a spam. You probably meant these logs that are being printed once you launch RMF in Humble or Iron in simulation with Gazebo:

```
[ign-21] [INFO] [1724119018.375995801] [lift_Lift1]: Reached floor L1 with doors closed
[ign-21] [INFO] [1724119019.167445240] [lift_Lift2]: Lift [Lift2] requested at level [L1]
[ign-21] [INFO] [1724119019.167494330] [lift_Lift2]: Reached floor L1 with doors closed
[ign-21] [INFO] [1724119019.388027364] [lift_Lift1]: Lift [Lift1] requested at level [L1]
```

In the RMF simulation code, the spam is being printed [here](https://github.com/open-rmf/rmf_simulation/blob/humble/rmf_building_sim_common/src/lift_common.cpp#L307-L315) and [here](https://github.com/open-rmf/rmf_simulation/blob/humble/rmf_building_sim_common/src/lift_common.cpp#L244-L246).

As explained by Grey [above](https://github.com/open-rmf/rmf_ros2/issues/315#issuecomment-1901895356), these lines are being printed because the lift supervisor continuously publishes `END_SESSION` requests when the lift is not being command by any robots. This was something we've found required during deployment.

Hence when you run this lift supervisor update with our simulation plugins, these lines will be printed as the simulated lifts are now receiving `END_SESSION` requests continuously. However in RMF Jazzy, there has been more conditions added to be checked before printing these lines (see code [here](https://github.com/open-rmf/rmf_simulation/blob/main/rmf_building_sim_gz_plugins/src/lift.cpp#L466-L473)). You'll notice that `Lift [Lift1] requested at level [L1]` will only be published once per unique lift request. I believe this is why you observed that the spam existed in Humble but not in Jazzy.

In Jazzy simulation, you will still see the following logs being printed during the lift travel:
```
[lift_supervisor-12] [INFO] [1724131849.880747544] [rmf_lift_supervisor]: [deliveryRobot/deliveryBot_1] Published lift request to [L3] from lift supervisor
[lift_supervisor-12] [INFO] [1724131849.881088925] [rmf_lift_supervisor]: [deliveryRobot/deliveryBot_1] Received adapter lift request to [L3] with request type [1]
[lift_supervisor-12] [INFO] [1724131849.890764576] [rmf_lift_supervisor]: [deliveryRobot/deliveryBot_1] Published lift request to [L3] from lift supervisor
[lift_supervisor-12] [INFO] [1724131849.890929346] [rmf_lift_supervisor]: [deliveryRobot/deliveryBot_1] Received adapter lift request to [L3] with request type [1]
```
These logs are from the lift supervisor [here](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/lift_supervisor/Node.cpp#L59-L63) and [here](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/lift_supervisor/Node.cpp#L103-L114) and are only printed when there is a lift request from the fleet adapters.

I hope I clarified a little about the difference in spam, but feel free to let me know if that wasn't clear!

xiyuoh, 2024-08-22:
Closing this ticket now as the the various issues observed seem to stem from other causes. Feel free to re-open if any bugs related to the continuous `END_SESSION` lift requests resurface.

closed 2024-08-22 by xiyuoh