open-rmf/rmf_ros2 #363: [Bug]: Typo/bug at convert_Graph.cpp (when entry_event is nullptr)

kjchee, 2024-06-07:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

main

### ROS distribution

Humble

### ROS installation type

Source build

### Description of the bug

1. There is a typo/bug at convert_Graph.cpp line 909 [https://github.com/open-rmf/rmf_ros2/blob/77c9f754e98f46462a0a4e47bb86f87cb48b002b/rmf_traffic_ros2/src/rmf_traffic_ros2/convert_Graph.cpp#L909](url): entry_event->execute(factory);

2. The entry_event here should be exit_event.

3. It crashed when entry_event is nullptr.

### Steps to reproduce the bug

1. in parse_graph.cpp, add a lane with exit event but without entry event, something like:
     
` const YAML::Node& lanes = level.second["lanes"];`
` for (const auto& lane : lanes)`
` {`
`    std::size_t begin = lane[0].as<std::size_t>();`
`    std::size_t end = lane[1].as<std::size_t>();`
`    rmf_utils::clone_ptr<Event> entry_event;`
`    rmf_utils::clone_ptr<Event> exit_event;` 
`    exit_event = Event::make(Lane::Dock(dock_name, duration));`
`    graph.add_lane(
           {begin, entry_event},
           {end, exit_event});`
` }` 

2. fleet adapter will crash at convert_Graph.cpp at line 909 [https://github.com/open-rmf/rmf_ros2/blob/77c9f754e98f46462a0a4e47bb86f87cb48b002b/rmf_traffic_ros2/src/rmf_traffic_ros2/convert_Graph.cpp#L909](url)

3. It crashed because the entry_event is nullptr.
4. So, the entry_event here should be exit_event.

mxgrey, 2024-06-07:
Excellent catch! You're absolutely right and #364 has been opened to fix this.

Luckily these serialized nav graphs were only being transmitted to the nav graph visualizer, so the serialized entry/exit events were never being used and weren't risking any misbehavior except for this crash. On the other hand, the lack of misbehavior is why this went for so long without being noticed.

luca-della-vedova, 2024-06-12:
#364 merged, closing this

closed 2024-06-12 by luca-della-vedova