open-rmf/rmf_ros2 #391: [Other issue]: Unable to compute a location on the navigation graph for robot [omni_base_1] being added to fleet [omni_fleet] using map [ground_floor] and position [19.982, -9.426, 0.236] specified in the initial_state argument. This can happen if the map in initial_state does not match any of the map names in the navigation graph supplied or if the position reported in the initial_state is far way from the navigation graph. This robot will not be added to the fleet.

tomkimsour, 2024-10-30:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

I am looking for instructions on how to fix the following issue the error raised by this message. 

```
Unable to compute a location on the navigation graph for robot [robot_1] being added to fleet [robot_fleet] using map [ground_floor] and position [19.982, -9.426, 0.236] specified in the initial_state argument. This can happen if the map in initial_state does not match any of the map names in the navigation graph supplied or if the position reported in the initial_state is far way from the navigation graph. This robot will not be added to the fleet.
```

mxgrey, 2024-10-30:
Take a look at the nav graph on `ground_floor` and check what vertices are close to the coordinates `{ x = 19.982, y = -9.426 }`. If there are no vertices reasonably close to those coordinates then either you're initializing your robot very far from your nav graph elements or there is something wrong with the transformation that you're using to convert between robot and RMF coordinates.

If you do find a vertex that's reasonably close to those coordinates then you may have a [`lane_merge_distance`](https://github.com/open-rmf/rmf_demos/blob/b1636b0de7d2bf951c613e3d5b9fc3bf46fc9a2b/rmf_demos/config/battle_royale/tinyRobot_config.yaml#L30) value that's too small (the default is 1.0m). Make sure the value for `lane_merge_distance` is essentially how far the robot is allowed to be from a nav graph element (lane or vertex) while still being allowed to use that element.

tomkimsour, 2024-10-30:
[Solved] To solve it I followed the following steps : 
- open traffic-editor
- add waypoint with is_charger and is_parking_slot
- add waypoint to graph by adding lanes
- regenerate nav_graph by compiling the project
- verify that nav_graph files contain the waypoint
- inside the flee_config.yaml update set the following : 
```

reference_coordinates:
  ground_floor:
    rmf: [[spawn_waypoint.x, spawn_waypoint.y]] # replace spawn_waypoint.x and spawn_waypoint.y by the waypoint coordinates from the nav_graph or traffic-editor.
    robot: [[0.0, 0.0]]

```

closed 2024-10-30 by tomkimsour