open-rmf/rmf_ros2 #415: [Bug]: Lift supervisor publishing even when lift is at destination floor

TanJunKiat, 2025-04-21:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 24.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

main

### ROS distribution

Jazzy

### ROS installation type

Source build

### Description of the bug

The current logic of the lift supervisor is that it will keep publishing even when the lift states (destination level and/or door state) arethe desired states.

https://github.com/open-rmf/rmf_ros2/blob/41e9f914342153b7010badd896d0a96579e92f67/rmf_fleet_adapter/src/lift_supervisor/Node.cpp#L102-L110

The behavior seems off as the request should only publish when there is a need to.

Is there a pair of missing brackets to encapsulate the `if` condition?

### Steps to reproduce the bug

nil

### Expected behavior

Lift request to only publish when lift states does not match desired states.

### Actual behavior

Lift request publishing even when lift states matches desired states.

mxgrey, 2025-04-21:
I think this is _somewhat_ related to this discussion around excessive lift supervisor activity: https://github.com/open-rmf/rmf/discussions/638 so I think it's worth linking the two together.

For the specific case that you're bringing up here, in the early days of RMF we published messages pretty aggressively, even if we had good reason to believe that the message had already been published, because we regularly ran into issues where DDS fails to deliver messages that we are depending on, even if we used reliable QoS.

If users are seeing that there's too much network traffic from this aggressive publishing, then we can consider an alternative strategy, but it may come at the cost of the current guarantee of [eventual consistency](https://en.wikipedia.org/wiki/Eventual_consistency), or it may require a more complex data flow that includes an ACK protocol.

TanJunKiat, 2025-04-21:
> For the specific case that you're bringing up here, in the early days of RMF we published messages pretty aggressively, even if we had good reason to believe that the message had already been [...]

Thinking out loud, does changing the `lift_requests` to a ROS2 service help with the DDS issue, since services requires a hand-shake between client and server.

Each lift adapter will be a server with a unique namepsace and the lift supervisor will request accordingly.

mxgrey, 2025-04-22:
> Thinking out loud, does changing the lift_requests to a ROS2 service help with the DDS issue, since services requires a hand-shake between client and server.

We specifically avoided using services in the early days of RMF because DDS-based services were consistently unreliable at that time. I'm not sure how much the situation has improved since then, but ultimately the reliability of services over a DDS-based middleware still depends on the QoS settings. Under the hood, ROS 2 services on a DDS middleware are still implemented using DDS pub/sub, so you'll need to have reliability turned on and you need to make sure your history buffer is large enough for all the service requests that you'll be sending. If the buffer isn't large enough for the number of agents requesting the lift, then some service requests can get dropped quietly, and there's no clear way to recover from that.

If we were using pure DDS we could take advantage of [keys](https://community.rti.com/static/documentation/connext-dds/current/doc/manuals/connext_dds_professional/getting_started_guide/csharp/intro_keys_instances.html), but for better or worse keys are not supported by ROS 2.

With rmw_zenoh being available, we're seriously considering a switch to action and service driven behaviors for the next generation, and Open-RMF users would be advised to use zenoh or else take responsibility for their DDS configuration to get reliable behavior. But I think we need to stick to pub/sub for the current generation, or else we introduce unexpected risks for users.

closed 2025-05-21 by TanJunKiat