open-rmf/rmf_ros2 #421: [Bug]: Incorrect Orientation Goal Check (Angle Wrap-around and Tolerance)

alex-roba, 2025-05-07:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

24.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

main

### ROS distribution

Rolling

### ROS installation type

Source build

### Description of the bug

The code checks if a robot's current orientation (l.orientation()) is significantly different from a target orientation derived from a waypoint (wp.position()[2]). This check is performed using the condition std::abs(wp.position()[2] - l.orientation()) * 180.0 / M_PI > 1e-2. 

https://github.com/open-rmf/rmf_ros2/blob/160176cb818b78559fafd8a8ad4af453ba6e4382/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/EasyFullControl.cpp#L1071

However, this approach presents two main problems that can lead to incorrect behavior:

First, the tolerance 1e-2 degrees (0.01 degrees) is extremely small. If the robot's actual orientation is, for instance, 179.7 degrees and the target is 180.0 degrees, the raw difference is 0.3 degrees. Since 0.3 is greater than 0.01, the condition is met, signaling that the orientation goal is not aligned. If 179.7 degrees is already within the robot's control or sensing precision relative to 180.0 degrees, subsequent commands might be sent to correct this tiny difference, but the robot may not actually move because it considers itself "on target" within its own internal tolerance. This can result in the system incorrectly believing the goal isn't met and potentially entering a loop or becoming stuck.

Second, and critically, the calculation std::abs(wp.position()[2] - l.orientation()) does not account for the inherent wrap-around nature of angles. The difference between two angles should represent the shortest angular distance between them (e.g., the difference between 179 degrees and -179 degrees is only 2 degrees, not 358 degrees). Using the absolute value of the raw difference means that if the target angle is just over the ±180 ∘
  boundary and the current angle is just under (or vice-versa), the calculated difference will be close to 360 degrees. This large, incorrect difference will always exceed the tiny tolerance, causing the condition to wrongly indicate that the orientation is far from the target when it is actually very close.

### Steps to reproduce the bug

launch

mxgrey, 2025-05-07:
Thanks for reporting this. We do have a [wrap_to_pi](https://github.com/open-rmf/rmf_utils/blob/main/rmf_utils/include/rmf_utils/math.hpp#L26-L35) function which is supposed to be used in cases like this, but it seems we forgot to use it here.

I think I can get a fix in for this tonight and include it in the upcoming release for kilted.

mxgrey, 2025-05-07:
> but the robot may not actually move because it considers itself "on target" within its own internal tolerance. This can result in the system incorrectly believing the goal isn't met and potentially entering a loop or becoming stuck.

I'm a bit confused by this issue. If the robot is on target then your integration code should just immediately report back to RMF that the robot has reached its destination. There will always be a risk that RMF will ask your robot to go to a destination that is with the robot's "already arrived" threshold. That risk comes from factors like asynchronicity and the arbitrary nature of thresholds for different nav systems. I would recommend making sure your integration code is robust to this kind of situation in general.

alex-roba, 2025-05-07:
yes my integration code is robust to this kind of situation in general and would just report back to RMF that the robot has reached its destination but because of strict tolerance the condition of https://github.com/open-rmf/rmf_ros2/blob/160176cb818b78559fafd8a8ad4af453ba6e4382/rmf_fleet_adapter/src/rmf_fleet_adapter/agv/EasyFullControl.cpp#L1179 will be met then I get the log" Requesting replan for [%s] because it is too far from its commanded path" then new command will be issued but my robot will report back to RMF that the robot has reached its destination and the RMF adapter is getting stuck in a replanning loop at waypoint

mxgrey, 2025-05-07:
Ah thanks for elaborating, now I understand why that's an issue.

I think the actual solution here is to not depend on the orientation at all when deciding if there's a valid "connection" to the planned path. Being sensitive to orientation wouldn't accomplish anything for that.

alex-roba, 2025-05-07:
Currently, I solved that by setting that strict tolerance to my robot tolerance but not sure if that is the best workaround or no if yes 
 I would suggest to parameterize that tolerance

alex-roba, 2025-05-07:
> Ah thanks for elaborating, now I understand why that's an issue.
> 
> I think the actual solution here is to not depend on the orientation at all when deciding if there's a valid "connection" to the planned path. Being sensitive to orientation wouldn't accomplish anything for that.

I believe that ensuring a reliable 'connection' based on orientation is crucial to maintain proper rotation for the setting of not to skip in place rotation.

mxgrey, 2025-05-07:
I think what we're seeing here reflects a conceptual flaw in how #398 was implemented. The purpose of that portion of the code is to ensure that the robot lies somewhere along the planned path. Checking if the robot's orientation matches various waypoints along that path is an arbitrary and unhelpful condition, because it means if a robot happens to be facing a different direction than what was planned, then we consider it to not be anywhere along the path and we need to replan it.

Instead to accommodate the not-skipping-rotations option, I think we should first find the waypoint along the path that where the robot is currently located, and then add in a rotation command post-hoc when needed.

closed 2025-06-24 by mxgrey