open-rmf/rmf_ros2 #424: [Bug]: `rmf_demos_gz` launch fails due to missing `rmf_reservation_node` package in branch **jazzy**

gaurishg, 2025-05-08:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu24.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

jazzy

### ROS distribution

Jazzy

### ROS installation type

Docker

### Package or library, if applicable

rmf_reservation_node

### Description of the bug

`rmf_reservation_*` packages are missing from the **jazzy** branch because of which `office.launch.xml` demo can not be launched.

### Steps to reproduce the bug

1. Copy the `rmf.repos` file to `~/ros2_ws` from `rmf` repo and `jazzy branch`.
2. `mkdir src && vcs import src < rmf.repos`
3. rosdep update and install
4. `colcon build --mixin release lld`
5. `source install/setup.bash`
6. `ros2 launch rmf_demos_gz office.launch.xml`

### Expected behavior

The office demo should be launched without any error or missing dependencies.

### Actual behavior

It fails with the following error log

```bash
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-08-02-02-27-444961-d637f33ce6e3-12298
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'rmf_reservation_node' not found, searching: [
  '/root/ros2_ws/install/rmf_workcell_msgs',
  '/root/ros2_ws/install/rmf_demos_gz',
  '/root/ros2_ws/install/rmf_demos',
  '/root/ros2_ws/install/rmf_fleet_adapter',
  '/root/ros2_ws/install/rmf_fleet_adapter_python',
  '/root/ros2_ws/install/rmf_task_ros2',
  '/root/ros2_ws/install/rmf_websocket',
  '/root/ros2_ws/install/rmf_visualization',
  '/root/ros2_ws/install/rmf_visualization_rviz2_plugins',
  '/root/ros2_ws/install/rmf_visualization_obstacles',
  '/root/ros2_ws/install/rmf_visualization_navgraphs',
  '/root/ros2_ws/install/rmf_visualization_floorplans',
  '/root/ros2_ws/install/rmf_visualization_fleet_states',
  '/root/ros2_ws/install/rmf_visualization_building_systems',
  '/root/ros2_ws/install/rmf_visualization_msgs',
  '/root/ros2_ws/install/rmf_traffic_ros2',
  '/root/ros2_ws/install/rmf_traffic_examples',
  '/root/ros2_ws/install/rmf_demos_bridges',
  '/root/ros2_ws/install/rmf_traffic_msgs',
  '/root/ros2_ws/install/rmf_traffic_editor_test_maps',
  '/root/ros2_ws/install/rmf_traffic_editor_assets',
  '/root/ros2_ws/install/rmf_traffic_editor',
  '/root/ros2_ws/install/rmf_demos_tasks',
  '/root/ros2_ws/install/rmf_task_msgs',
  '/root/ros2_ws/install/rmf_task_sequence',
  '/root/ros2_ws/install/rmf_task',
  '/root/ros2_ws/install/rmf_battery',
  '/root/ros2_ws/install/rmf_utils',
  '/root/ros2_ws/install/rmf_robot_sim_common',
  '/root/ros2_ws/install/rmf_building_map_tools',
  '/root/ros2_ws/install/rmf_site_map_msgs',
  '/root/ros2_ws/install/rmf_scheduler_msgs',
  '/root/ros2_ws/install/rmf_robot_sim_gz_plugins',
  '/root/ros2_ws/install/rmf_fleet_msgs',
  '/root/ros2_ws/install/rmf_building_sim_gz_plugins',
  '/root/ros2_ws/install/rmf_lift_msgs',
  '/root/ros2_ws/install/rmf_ingestor_msgs',
  '/root/ros2_ws/install/rmf_door_msgs',
  '/root/ros2_ws/install/rmf_dispenser_msgs',
  '/root/ros2_ws/install/rmf_demos_maps',
  '/root/ros2_ws/install/rmf_demos_assets',
  '/root/ros2_ws/install/rmf_charger_msgs',
  '/root/ros2_ws/install/rmf_charging_scheme_msgs',
  '/root/ros2_ws/install/rmf_building_map_msgs',
  '/root/ros2_ws/install/rmf_api_msgs',
  '/root/ros2_ws/install/pybind11_json_vendor',
  '/root/ros2_ws/install/nlohmann_json_schema_validator_vendor',
  '/root/ros2_ws/install/ament_cmake_catch2',
  '/opt/ros/jazzy'
]"
```

xiyuoh, 2025-05-08:
As mentioned in https://github.com/open-rmf/rmf_demos/issues/298#issuecomment-2861866472 where I replied to the wrong ticket, some updates from `main` were mistakenly ported into the `jazzy` branch. The reservations feature is not meant to be on Jazzy. I've just [removed](https://github.com/open-rmf/rmf_demos/commit/97e17a83bee0d2a0f298fbae1965efdf4dfd050a) the reservations-related params on `jazzy` to rectify this, but still testing to make sure that the branch works as it should.

xiyuoh, 2025-05-08:
@gaurishg After some testing and fixes, the `jazzy` branch of rmf_demos should be working now. Let me know if things don't work for you, otherwise feel free to close this ticket.

mxgrey, 2025-05-08:
Note that to update your local copy of the `jazzy` branch, you will need to do 

```bash
git fetch
```

```
git checkout jazzy
```

```
git reset --hard origin/jazzy
```

assuming your `origin` is this repo.

gaurishg, 2025-05-09:
Thanks @xiyuoh, Gazebo and Rviz successfully showed up. I see a lot of warnings in the terminal related to gazebo but that does not seem to affect any functionality.

[image]

closed 2025-05-09 by gaurishg