open-rmf/rmf_ros2 #429: [Other issue]: rmf_charging_schedule not released on RHEL

Yadunund, 2025-05-12:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

The [python3-icecream](https://github.com/open-rmf/rmf_ros2/blob/f96c074c0b0ce6b5f658a7ee0ca84ca02e8ab39a/rmf_charging_schedule/package.xml#L12) is not available on RHEL so RPMs for this package are not released.

mxgrey, 2025-05-12:
Is it possible for us to add it to rosdistro? It's an incredibly useful utility that I think the ROS python users will benefit from.

Yadunund, 2025-05-13:
It's in rosdistro for Debian & Ubuntu platforms and hence build succeeds for those platforms. But a key for RHEL is missing. I haven't checked if it's distributed via EPEL...

https://github.com/ros/rosdistro/blob/673b6e227492b362d14f7163f8ea158b1c04be3c/rosdep/python.yaml#L6702

mxgrey, 2025-05-13:
Oh I see... The important functionality of the node doesn't depend on icecream, so I'll try to work out a way to get rid of it.

Yadunund, 2025-05-20:
Closing as the dep was removed https://github.com/open-rmf/rmf_ros2/pull/432

closed 2025-05-20 by Yadunund