open-rmf/rmf_ros2 #434: [Bug]: Segmentation Fault in RMF Adapter when instantiating a Transformation Object

AscendedSky, 2025-05-20:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

WSL Ubuntu 22.04

### Open-RMF installation type

Binaries

### Open-RMF version or commit hash

release-jazzy-240617

### ROS distribution

Jazzy

### ROS installation type

Source build

### Package or library, if applicable

rmf_fleet_adapter

### Description of the bug

When trying out free fleet, the fleet adapter process crashes with a segmentation fault when returning the newly instantiated `Transformation` object in the helper function `compute_transforms` in the file `fleet_adapter.py`. This seems to be an issue with the underlying rmf_core. 

```
(ros_env) mayukh_ghosh@LAPTOP-DN9HAK9L:~/ff_ws$ ros2 launch free_fleet_examples nav2_tb3_simulation_fleet_adapter.launch.xml
[INFO] [launch]: All log files can be found below /home/mayukh_ghosh/.ros/log/2025-05-20-21-49-46-001850-LAPTOP-DN9HAK9L-41127
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fleet_adapter.py-1]: process started with pid [41130]
[fleet_adapter.py-1] [retreat_to_charger_interval] value is not provided, default to 10 seconds
[fleet_adapter.py-1] Finishing request: nothing
[fleet_adapter.py-1] Fleet is not configured to perform any finishing request
[fleet_adapter.py-1] Maximum delay is not provided, default to 10.0
[fleet_adapter.py-1] [INFO] [1747748987.080688243] [turtlebot3_fleet_adapter]: Parameter [discovery_timeout] set to: 60.000000
[fleet_adapter.py-1] [INFO] [1747748987.094496337] [turtlebot3_fleet_adapter]: Registering to query topic rmf_traffic/query_update_1
[fleet_adapter.py-1] [INFO] [1747748987.105833599] [turtlebot3_fleet_adapter]: Mirror handling new sync of 1 queries from schedule node [aae8d10d-8091-4c44-af9f-bba8a96b025c]
[fleet_adapter.py-1] [INFO] [1747748988.118790707] [turtlebot3_command_handle]: Transformation error estimate for L1: 0.0009797264984120714
[ERROR] [fleet_adapter.py-1]: process has died [pid 41130, exit code -11, cmd '/home/mayukh_ghosh/ff_ws/install/free_fleet_adapter/lib/free_fleet_adapter/fleet_adapter.py -c /home/mayukh_ghosh/ff_ws/install/free_fleet_examples/share/free_fleet_examples/config/fleet/nav2_tb3_simulation_fleet_config.yaml -n /home/mayukh_ghosh/ff_ws/install/free_fleet_examples/share/free_fleet_examples/maps/turtlebot3_world/nav_graphs/0.yaml -s --ros-args --params-file /tmp/launch_params_nrk9pygx --params-file /tmp/launch_params_zzp92b_d'].
```
Here is the `gdb` backtrace:

```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffd48df849 in ?? ()
   from /opt/ros/jazzy/lib/python/site-packages/rmf_adapter.cpython-312-x86_64-linux-gnu.so
#2  0x00007fffd48d4156 in ?? ()
   from /opt/ros/jazzy/lib/python/site-packages/rmf_adapter.cpython-312-x86_64-linux-gnu.so
#3  0x00007fffd489e759 in ?? ()
   from /opt/ros/jazzy/lib/python/site-packages/rmf_adapter.cpython-312-x86_64-linux-gnu.so
#4  0x000000000058208f in ?? ()
#5  0x0000000000549185 in _PyObject_MakeTpCall ()
#6  0x000000000054cea7 in ?? ()
#7  0x000000000059e231 in ?? ()
#8  0x0000000000599b63 in ?? ()
#9  0x00007fffd6022379 in ?? ()
   from /opt/ros/jazzy/lib/python3.12/site-packages/rclpy/_rclpy_pybind11.cpython-312-x86_64-linux-gnu.so
#10 0x0000000000549185 in _PyObject_MakeTpCall ()
#11 0x00000000005d73c9 in _PyEval_EvalFrameDefault ()
#12 0x00000000005d58eb in PyEval_EvalCode ()
#13 0x0000000000608b42 in ?? ()
#14 0x00000000006b4e93 in ?? ()
#15 0x00000000006b4bfa in _PyRun_SimpleFileObject ()
#16 0x00000000006b4a2f in _PyRun_AnyFileObject ()
#17 0x00000000006bca95 in Py_RunMain ()
#18 0x00000000006bc57d in Py_BytesMain ()
#19 0x00007ffff7c831ca in __libc_start_call_main (main=main@entry=0x5189b0, argc=argc@entry=12,
    argv=argv@entry=0x7fffffffcb58) at ../sysdeps/nptl/libc_start_call_main.h:58
#20 0x00007ffff7c8328b in __libc_start_main_impl (main=0x5189b0, argc=12, argv=0x7fffffffcb58,
--Type <RET> for more, q to quit, c to continue without paging--
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffcb48)
    at ../csu/libc-start.c:360
#21 0x0000000000657ce5 in _start ()
```
Please do let me know if any other information is required. I would also appreciate any "hacky" workarounds to help resolve this issue. 

### Steps to reproduce the bug

1. ros2 launch free_fleet_examples turtlebot3_world_rmf_common.launch.xml 
2. ros2 launch free_fleet_examples nav2_tb3_simulation_fleet_adapter.launch.xml

There is no need to run any of the other commands given in the free fleet README.

xiyuoh, 2025-05-20:
Hi @AscendedSky !

There was another [ticket](https://github.com/open-rmf/fleet_adapter_template/issues/34#issuecomment-2880124066) that looked in-depth into possible segfaults caused by the Transformation object, but we have not been able to reproduce the crash, and the user eventually found that it only occurs when a specific combination of Python packages are installed. Their error logs look exactly the same as yours, so I'd recommend taking a look and see if you can find anything useful there. If there are new clues pointing to the crash originating from RMF libraries, we'd really appreciate if you can share and/or open PRs for a fix.

Other than that, I've been able to run free fleet successfully without crashing with Jazzy binaries installed via `sudo apt install ros-jazzy-rmf-dev` on Linux. I see that you're using WSL Ubuntu 24.04; would you be able to try installing RMF Jazzy binaries following [these instructions](https://github.com/open-rmf/rmf/tree/release-jazzy-240617?tab=readme-ov-file#binary-installation)? Instead of downloading the release tar.gz. I'm interested to see if it makes a difference, and narrow down what's causing the crash.

AscendedSky, 2025-05-20:
Hello @xiyuoh . Thanks for the prompt reply. I am not too sure what you mean by installing the Jazzy Binaries since the instructions are the same as the one in main. However, I have tried to build RMF from source, where I managed to find this `gdb` backtrace:

```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffd488fec8 in pybind11::detail::type_caster<Eigen::Matrix<double, 2, 1, 0, 2, 1>, void>::load (
    this=0x7fffffff92d0, src=..., convert=true) at /usr/include/pybind11/eigen/matrix.h:327
#2  0x00007fffd488fb38 in pybind11::detail::argument_loader<pybind11::detail::value_and_holder&, double, double, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::load_impl_sequence<0ul, 1ul, 2ul, 3ul> (this=0x7fffffff92d0,
    call=...) at /usr/include/pybind11/detail/../cast.h:1465
#3  0x00007fffd488f6fd in pybind11::detail::argument_loader<pybind11::detail::value_and_holder&, double, double, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::load_args (this=0x7fffffff92d0, call=...)
    at /usr/include/pybind11/detail/../cast.h:1443
#4  0x00007fffd488f559 in _ZZN8pybind1112cpp_function10initializeIZNS_6detail8initimpl11constructorIJddN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEE7executeINS_6class_IN17rmf_fleet_adapter3agv14TransformationEJEEEJNS_3argESF_SF_ETnNSt9enable_ifIXntsrT_9has_aliasEiE4typeELi0EEEvRSH_DpRKT0_EUlRNS2_16value_and_holderEddS7_E_vJSQ_ddS7_EJNS_4nameENS_9is_methodENS_7siblingENS2_24is_new_style_constructorESF_SF_SF_EEEvOSH_PFT0_DpT1_EDpRKT2_ENKUlRNS2_13function_callEE_clES17_ (this=0x7fffffff933f, call=...) at /usr/include/pybind11/pybind11.h:228
#5  0x00007fffd488f519 in _ZZN8pybind1112cpp_function10initializeIZNS_6detail8initimpl11constructorIJddN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEE7executeINS_6class_IN17rmf_fleet_adapter3agv14TransformationEJEEEJNS_3argESF_SF_ETnNSt9enable_ifIXntsrT_9has_aliasEiE4typeELi0EEEvRSH_DpRKT0_EUlRNS2_16value_and_holderEddS7_E_vJSQ_ddS7_EJNS_4nameENS_9is_methodENS_7siblingENS2_24is_new_style_constructorESF_SF_SF_EEEvOSH_PFT0_DpT1_EDpRKT2_ENUlRNS2_13function_callEE_8__invokeES17_ (call=...) at /usr/include/pybind11/pybind11.h:224
#6  0x00007fffd47741fd in pybind11::cpp_function::dispatcher (self=0x7fffd619ccf0, args_in=0x7fffb1ce6e30,
    kwargs_in=0x0) at /usr/include/pybind11/pybind11.h:946
#7  0x000000000058208f in ?? ()
#8  0x0000000000549185 in _PyObject_MakeTpCall ()
#9  0x000000000054cea7 in ?? ()
#10 0x000000000059e231 in ?? ()
#11 0x0000000000599b63 in ?? ()
--Type <RET> for more, q to quit, c to continue without paging--
#12 0x00007fffd476c413 in pybind11_meta_call (type=0x1599730, args=0x7fffb1c97380, kwargs=0x0)
    at /usr/include/pybind11/detail/class.h:187
#13 0x0000000000549185 in _PyObject_MakeTpCall ()
#14 0x00000000005d73c9 in _PyEval_EvalFrameDefault ()
#15 0x00000000005d58eb in PyEval_EvalCode ()
#16 0x0000000000608b42 in ?? ()
#17 0x00000000006b4e93 in ?? ()
#18 0x00000000006b4bfa in _PyRun_SimpleFileObject ()
#19 0x00000000006b4a2f in _PyRun_AnyFileObject ()
#20 0x00000000006bca95 in Py_RunMain ()
#21 0x00000000006bc57d in Py_BytesMain ()
#22 0x00007ffff7c831ca in __libc_start_call_main (main=main@entry=0x5189b0, argc=argc@entry=13,
    argv=argv@entry=0x7fffffffa4e8) at ../sysdeps/nptl/libc_start_call_main.h:58
#23 0x00007ffff7c8328b in __libc_start_main_impl (main=0x5189b0, argc=13, argv=0x7fffffffa4e8,
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffa4d8)
    at ../csu/libc-start.c:360
#24 0x0000000000657ce5 in _start ()
```
After confirming that ref.ptr(), and buf.ptr() were not null pointers and pointed to correctly dimensioned arrays, I have found that the issue lies with NumPy itself instead of a combination of packages. My NumPy version was 2.2.6, and after downgrading to 1.26.0, the issue was solved. I have not tested it with binaries yet, but I believe it should work. Thanks once again.

xiyuoh, 2025-05-21:
Thanks for clarifying! I saw that your RMF version was `release-jazzy-240617`, so I assumed you installed it from the [release page](https://github.com/open-rmf/rmf/releases) rather than `apt install`. Good to know it's the same.

This is a great find, thanks for narrowing down the exact versioning issues caused by numpy, it is really helpful! I'll raise this to the other user who faced similar issues. I'll close this ticket for now, feel free to reopen if things don't work for you on binaries.

closed 2025-05-21 by xiyuoh