open-rmf/rmf_ros2 #448: [Bug]: Title: Unexpected deadlock caused by Mutex

suztakur, 2025-06-20:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 24.04

### Open-RMF installation type

Source build

### Other Open-RMF installation methods

https://github.com/open-rmf/rmf

### Open-RMF version or commit hash

main-12b5980859562a11cb4358c0dc0e34fb9fcec7a6

### ROS distribution

Jazzy

### ROS installation type

Source build

### Package or library, if applicable

rmf_fleet_adapter

### Description of the bug

We have confirmed the occurrence of a deadlock due to an unexpected Mutex lock waiting state.
Mutex is set as shown in the following attached image for the map in the nearby area where the problem occurred.
Please check rmf_demos/rmf_demos_maps/maps/test/test.building.yaml in the attached demo package for details of the map.

[image]

Rviz images of the problem are shown below.

[image]

Each robot is submitting a task that should be moving downward.
At this time, strangely enough, tinyRobot/tinyRobot5 is trying to lock tinyRobot5_charger_mutex again, even though it has already left.
At this time, a deadlock occurs because tinyRobot/tinyRobot4 has already acquired tinyRobot5_charger_mutex.

```
[fleet_adapter-17] [INFO] [1750310896.033224395] [tinyRobot_fleet_adapter]: [tinyRobot/tinyRobot5] is waiting to lock mutex group [tinyRobot5_charger_mutex] but that mutex is currently held by [tinyRobot/tinyRobot4]
```

It is also strange that the yellow marker is shifted from the robot's position.
I would like to ask your advice on whether I should modify the code or the NavigationGraph.
I am also attaching the Rosbag as it appeared when the problem occurred.

[rosbag2_2025_06_19-14_24_42.zip](https://github.com/user-attachments/files/20829096/rosbag2_2025_06_19-14_24_42.zip)

### Steps to reproduce the bug

1.[rmf_repos.zip](https://github.com/user-attachments/files/20829105/rmf_repos.zip) is used to build the RMF environment.
2.replace the demonstrations repository.
3.RMF launch
```
ros2 launch rmf_demos_gz test.launch.xml
```
4.Task submission
```
ros2 launch rmf_demos test_loop.launch.xml
```

### Expected behavior

Each robot runs without deadlock caused by Mutex.

### Actual behavior

Deadlock is caused by Mutex.

suztakur, 2025-06-20:
Additional information.
After analyzing the attached rosbag, we have identified a strange behavior.
Checking the /rmf_traffic/itinerary_set topic when the yellow marker transitions, we found that the path to the coordinates to the yellow marker is output.
Usually, the route should be output from tinyRobot5's current robot position to the downward direction where it is moving to.
We think that the output of a route that is not necessary is causing a deadlock.

[image]

mxgrey, 2025-06-20:
> Open-RMF version or commit hash
> 
> main-12b5980859562a11cb4358c0dc0e34fb9fcec7a6

This appears to be the commit hash of https://github.com/open-rmf/rmf which doesn't tell us much about what version of the implementation you have running. Could you share the commit hash of the https://github.com/open-rmf/rmf_ros2 repo that you're using so we can make sure to test against the correct version?

> We think that the output of a route that is not necessary is causing a deadlock.

The decision about when to lock a mutex group happens in [`ExecutePlan.cpp`](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/rmf_fleet_adapter/events/ExecutePlan.cpp) in these locations: [[1](https://github.com/open-rmf/rmf_ros2/blob/d3f6bcf25267f9adaf9a41f074a919b6292300c8/rmf_fleet_adapter/src/rmf_fleet_adapter/events/ExecutePlan.cpp#L1009-L1053)][[2](https://github.com/open-rmf/rmf_ros2/blob/d3f6bcf25267f9adaf9a41f074a919b6292300c8/rmf_fleet_adapter/src/rmf_fleet_adapter/events/ExecutePlan.cpp#L194-L201)]. It's based entirely on whether the most recent plan will bring the robot through a waypoint or lane with a mutex group [[1](https://github.com/open-rmf/rmf_traffic/blob/251cbee41fc6448d8ad3ad9ec94faa77faa6e933/rmf_traffic/include/rmf_traffic/agv/Graph.hpp#L206)][[2](https://github.com/open-rmf/rmf_traffic/blob/251cbee41fc6448d8ad3ad9ec94faa77faa6e933/rmf_traffic/include/rmf_traffic/agv/Graph.hpp#L585)].

Most likely there is something causing the planner to think it needs to backtrack up that lane or that it is still "on" the lane and therefore still needs to have the mutex group locked. Is this using the typical `rmf_demos_fleet_adapter` simulation stack?

mxgrey, 2025-06-20:
Since that's a brand new commit, could you check whether you still observe the problem on the [previous commit](https://github.com/open-rmf/rmf_ros2/commit/73b086fb6c5d2787e84f1ecda0dd30ebee1f7545). If the problem goes away after rolling back then that would indicate the problem is in those recent changes.

mxgrey, 2025-09-21:
Sorry for how long it took me to get back to this problem.

I've taken some time to remind myself of how this code works. The logic is rather convoluted, which is why it took some time.

The reason it wasn't done the way you've edited it is because whenever `mutex_group_change` is set to `true` in that lambda, we will end up inserting a [`LockMutexGroup`](https://github.com/open-rmf/rmf_ros2/blob/main/rmf_fleet_adapter/src/rmf_fleet_adapter/events/LockMutexGroup.hpp) event. That event would have the robot stop unnecessarily if we're only releasing a mutex and not gaining a new one.

It's possible that there's a logic error somewhere in this execution code, although it's still not jumping out to me. Another possibility is that you were being affected by [this bug](https://github.com/open-rmf/rmf_ros2/pull/473) which has been fixed in the latest `main` branch. It might be worth trying that branch to see if it fixes the problem you were experiencing. Otherwise I'll try to keep examining where the flaw is in the execution logic.

aaronchongth, 2025-11-18:
@mxgrey have we been able to reproduce this?